OpenTabletDriver/OpenTabletDriver #2974: Digimend drivers conflicts with OpenTabletDriver

gonX, 2023-10-21:
Digimend runs [hid-rebind](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/hid-rebind) on [udev hid device add](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/udev.rules)(?) which results in our module blacklist seemingly not working.

The other-driver-detection fix is likely to check for the presence of this `hid-rebind` script.

Packagers/maintainers can help work around this by marking the OpenTabletDriver package as incompatible with any relevant Digimend packages that provide this udev+script behavior.

I have personally not played around with the Digimend drivers since installing OTD, [but another user on the Discord](https://discord.com/channels/615607687467761684/723399565780189234/1165328637625638922) recently brought it to our attention that running `make install` with digimend results in OTD not being able to detect their Gaomon PD1161.

It is worth noting that it is correctly detected by the driver detection routine as UCLogic, which IMO trivializes support for this issue, making it a low priority at the moment.

e-Wulffe, 2023-10-29:
Hi, this is my first time posting within the issues of a project so am sorry for any missteps. I have been using the digimend drivers for the past 3 years, ever since I started using linux recently I came across OTD through a mention on another FOSS project. I decided to give it a shot just to see what it (and its plugins) were like. 

I am running XFCE on endeavouros and had my tablet, a Huion Kamvas Pro 16 (GT-156), operating fine with the digimend kernel drivers (don't know about the buttons or touch strip as I don't use them so didn't bother configuring them). I installed the AUR OTD binary package but it failed to recognise the tablet, I then uninstalled the digimend kernel drivers and the xf86-input-wacom package, in case there was a conflict, and rebooted. Still OTD failed to recognise the tablet. I do have an alternative functioning setup with digimend I can go back to but since currently none of my projects need me to make use of a pen display I would like to see if I can get this up and running

If I have made a mistake with regards to the installation or setup I am very sorry for troubling everyone over them.  Also  please let me know if this is not the right place to be bringing this us up or if any logs or outputs are needed. Thanks to anyone who takes the time to read this.

jamesbt365, 2023-10-29:
@e-Wulffe this is not a support request issue, and if your tablet isn't the 2.5k variant its not supported. Please see the supported tablet list on our website.

e-Wulffe, 2023-10-29:
Thanks for the prompt reply, I was confused because in the missing features section of the list it mentions the Kamvas Pro 16 without any other specifications (resolution, model number, year, etc), it might be worthwhile adding that in if there are only particular tablets bearing that name that qualify for the aforementioned partial support. Regarding my issue, I don't know if support for older tablets is being added, if it is I can create a new issue for it.

gonX, 2023-10-29:
@e-Wulffe Yes, please make a new issue using the Configuration Request

e-Wulffe, 2023-10-29:
Will do, thank you for your help.