OpenTabletDriver/OpenTabletDriver #2977: Support Request: H420X Not Detected

minisat0shi, 2023-10-22:
### What do you need help with?

H420X is said to be supported, however the application does not detect it. I've run through the issues I've found here and nothing seems to get it to work. Attempted to run on 5.3.3, but cannot get 5.3.3 to run - errors out stating I need to install .NET (of which I have v6 and v7 installed).

Happy to provide whatever info I can.

### What have you tried?

Uninstalled all drivers and used the driver cleanup tool. Uninstalled all drivers also from device manager. 

### Diagnostic Information

[diogs.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13062086/diogs.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-10-22:
Fixed in #2938.

>  Attempted to run on 5.3.3, but cannot get 5.3.3 to run - errors out stating I need to install .NET (of which I have v6 and v7 installed).

Of which you don't have v5 installed. We also do not provide support for 0.5.x anymore, so do not use it.

closed 2023-10-22 by jamesbt365

minisat0shi, 2023-10-22:
No, I have v 6.3 installed. Zip from github 6.3.

minisat0shi, 2023-10-22:
The 5.3 I mentioned does not even open. That was only a test

jamesbt365, 2023-10-22:
> No, I have v 6.3 installed. Zip from github 6.3.

I know. I also never said you were using 0.5.3.3. I said you **didnt have** dotnet 5 which is required for 0.5.3.3 and I also said not to use it, not that you were.

> The 5.3 I mentioned does not even open. That was only a test

Please read my comments carefully and refrain from opening more issues.

You can either follow the pull request I linked prior or wait until the next release of OTD.

minisat0shi, 2023-10-22:
My apologies, been a long day clearly my reading comprehension is suffering!

How can I implement the change from the linked pull request? Thanks!

jamesbt365, 2023-10-22:
[H420X.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13063041/H420X.json.txt)

remove the .txt off the end of this file, create a folder called Configurations next to the driver and place this file into it.