OpenTabletDriver/OpenTabletDriver #2984: Cursor randomly freezing for 0.5s

Leclowndu93150, 2023-10-28:
### Steps to Reproduce

https://github.com/OpenTabletDriver/OpenTabletDriver/assets/73310949/d8bbb084-335f-40b7-8631-003bbb35ccba

so many plays of mine ruined (this issue has been going for a year now) even tablet changes didn't do anything

Windows 10 22h2 btw

### Expected Behavior

work fine

### Observed Behavior

cursor teleporting

### Tablet Device

Wacom CTL-472 | XP Pen G430s (same issue)

### Diagnostic Information

https://pastes.dev/w1tBiECXFH

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-10-28:
what cpu?

Leclowndu93150, 2023-10-28:
> what cpu?

i5 11400H

Leclowndu93150, 2023-10-28:
[DxDiag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13196074/DxDiag.txt)

jamesbt365, 2023-10-28:
check your cpu usage during gameplay, the only time OTD should ever "freeze" is because it is lacking cpu time.

Leclowndu93150, 2023-10-28:
ah how do i do that

Leclowndu93150, 2023-10-28:
cpu spiked to 27% from 17% during that freeze

Tyfaspo, 2023-11-17:
i had similar experience with OTD but not with hawku, when using windows, switched to linux and since then this problem never occurred

gonX, 2023-11-19:
This is sadly help to hard diagnose as it can be caused by many things ranging from an underpowered hardware, incorrectly configured operating system, a dying tablet, random electromagnetic interference or a combination of the above.

Are you using a modded version of Windows, or any other power user improvements?

I would first try to see if you can reproduce this on a freshly formatted Windows install that _only_ has graphics drivers, OpenTabletDriver (and any needed dotnet runtimes) and osu! installed.

Leclowndu93150, 2023-11-20:
> This is sadly hard to diagnose as it can be caused by many things ranging from underpowered hardware, incorrectly configured operating system, a dying tablet, random electromagnetic interference or a combination of the above.
> 
> Are you using a modded version of Windows, or any other power user improvements?
> 
> I would first try to see if you can reproduce this on a freshly formatted Windows install that _only_ has graphics drivers, OpenTabletDriver (and any needed dotnet runtimes) and osu! installed.

a windows clean install fixed the issue so i guess it was performance issue

closed 2023-11-20 by jamesbt365