OpenTabletDriver/OpenTabletDriver #2987: Add support for Huion Kamvas Pro 16 (GT-156) <Linux> <X11> <Arch>

e-Wulffe, 2023-10-29:
### Manufacturer

Huion

### Product Name

GT-156

### Pressure Levels

8192

### Additional Information

- Works with digimend kernel drivers.
- Running EndeavourOS XFCE

### Diagnostic Information

[otd_diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13198763/otd_diagnostics.txt)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-10-29:
Before I check if a new configuration is actually needed could you set up OTD properly by following the FAQ page here? Thanks.

https://opentabletdriver.net/Wiki/FAQ/Linux#argumentoutofrangeexception

jamesbt365, 2024-08-08:
Closing as there is not the required information provided to verify if a configuration is needed, let alone create one.

closed 2024-08-08 by jamesbt365