OpenTabletDriver/OpenTabletDriver #2988: Support Request: OpenTabletDriver won't detect Gaomon S620

VersilverDeveloper, 2023-10-30:
### What do you need help with?

OpenTabletDriver can't find Gaomon S620. It says that a Gaomon Tablet Driver has been found yet can't find the tablet itself.

### What have you tried?

I tried:
Plugging and un-plugging
Restarting the PC
Reinstalling every driver
Clearing my PC from other tablet drivers
Reading other issue forum thing, didn't help me somehow.

https://opentabletdriver.net/Wiki FAQ

### Diagnostic Information

[diagnostics_versilvergaomon.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13209128/diagnostics_versilvergaomon.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-10-30:
https://github.com/X9VoiD/TabletDriverCleanup/releases/latest

Duplicate of many issues related to other drivers being installed.

closed 2023-10-30 by jamesbt365

VersilverDeveloper, 2023-10-30:
uhhh i did that

jamesbt365, 2023-10-30:
Did you say yes to everything that popped up when you ran it as admin? Because if you just ran the program and then didn't say yes then its not going to do anything.

VersilverDeveloper, 2023-10-30:
I decided to leave the WinUSB driver because the download guide tells me to use Zadig and change current drivers to WinUSB. I also forgot to mention that without the Gaomon driver the tablet just can't do anything (and for some reason it just wont launch now the moment i reinstalled the Gaomon driver)

jamesbt365, 2023-10-30:
> I decided to leave the WinUSB driver because the download guide tells me to use Zadig and change current drivers to WinUSB.

If you look again, it says **don't** install WinUSB for tablets that do not need it, and it doesn't say the Gaomon S620 needs it.

>I also forgot to mention that without the Gaomon driver the tablet just can't do anything (and for some reason it just wont launch now the moment i reinstalled the Gaomon driver)

This will be resolved by uninstalling WinUSB and the gaomon driver.

Read the instructions carefully next time, thanks!

VersilverDeveloper, 2023-10-30:
Oh alright, I will try that