OpenTabletDriver/OpenTabletDriver #2991: Pen rotation is inverted

sezanzeb, 2023-11-06:
### Description

My tilt is incorrect, it causes to rotate the other way around.

I first assumed this was because my tablet is rotated, however, it seems to be also the case when the rotation setting in the gui is set to 0°

To compensate this, I can set this in krita:

[image]

which makes it work perfectly fine.

---

I then tried to extend [input-remapper](https://github.com/sezanzeb/input-remapper) to invert the tilt, but the tilt stayed fixed in one position. Du you happen to know if something with the capabilities below is wrong, so that my system won't detect it as stylus?

```
Select the device event number [0-25]: 24
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x1 product 0x1 version 0x1
Input device name: "input-remapper stylus"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 321 (BTN_TOOL_RUBBER)
    Event code 329 (BTN_STYLUS3)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
    Event code 332 (BTN_STYLUS2)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min   -32768
      Max    32768
    Event code 1 (ABS_Y)
      Value      0
      Min   -32768
      Max    32768
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max    32768
    Event code 26 (ABS_TILT_X)
      Value      0
      Min     -127
      Max      127
    Event code 27 (ABS_TILT_Y)
      Value      0
      Min     -127
      Max      127
  Event type 21 (EV_FF)
Properties:
Testing ... (interrupt to exit)
Event: time 1699304497.782782, type 3 (EV_ABS), code 26 (ABS_TILT_X), value -1
Event: time 1699304497.782782, -------------- SYN_REPORT ------------
Event: time 1699304497.782880, type 3 (EV_ABS), code 27 (ABS_TILT_Y), value -51
Event: time 1699304497.782880, -------------- SYN_REPORT ------------
Event: time 1699304497.786364, type 3 (EV_ABS), code 26 (ABS_TILT_X), value 11
Event: time 1699304497.786364, -------------- SYN_REPORT ------------
Event: time 1699304497.790307, type 3 (EV_ABS), code 27 (ABS_TILT_Y), value -57
```

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2023-11-06:
Huion?

sezanzeb, 2023-11-06:
oh sorry, forgot to mention.

Yes, a huion camvas 16 2021

jamesbt365, 2023-11-06:
Duplicate of #2378.

closed 2023-11-06 by jamesbt365

sezanzeb, 2023-11-06:
oh sorry, I wonder how I was able to miss that when searching the issues. Thanks

Anyhow, do you have an idea what might be wrong with the capabilities posted above?

jamesbt365, 2023-11-06:
Parser is wrong, needs invert, see linked issue.

sezanzeb, 2023-11-06:
No I mean, I have my own uinput that I attempted to use to transform the tilt events. The capabilities above are not from opentabletdriver