OpenTabletDriver/OpenTabletDriver #2992: Incorrect auxiliary button behavior for Wacom CTE-440

LostInSwiss, 2023-11-07:
### Steps to Reproduce

The Wacom CTE-440 has three auxiliary buttons on the tablet. The middle button is a scroll button. They almost work but not quite.

In the tablet debugger the last two values of the raw tablet data relate to these buttons but the behavior of the tablet report Aux Buttons is incorrect and cannot be used for binding.

Below is a table of the button, the last two hex digits of the tablet data, the mapping with the button and what is reported in AuxButtons and a comment on the strange behavioure.

Button, Hex value, AuxButtons
1 left button, 40,  [False,False,False,True] - This is the only button that works correctly. True when pressed, all clear to false when depressed
2 scrl up = 08, [True, False, False, False] - This sort of works but does not clear to FFFF when depressed
3 scrl down = 38, [True, True, True, False] - This incorrectly sets three buttons to true and does not clear them when depressed 
4 right button = 80, This button will only clear all values to False if one is stuck on True, it will not set any buttons to true.

The issue is inputs do not set buttons to true and one button sets three buttons to true. The resulting behavior is only the left aux button works correctly, scroll up only fires a button once (until cleared by another), scroll down fires three buttons at once (and wont clear on depress), and the right button does nothing at all.

While the UI does not show 4 aux buttons (it should) I was able to add 2 more by editing the json settings file.

### Expected Behavior

Each button should trigger a single aux button when pressed and clear the aux buttons to false when depressed.

### Observed Behavior

The issue is inputs do not set buttons to true and one button sets three buttons to true. The resulting behavior is only the left aux button works correctly, scroll up only fires a button once (until cleared by another), scroll down fires three buttons at once (and wont clear on depress), and the right button does nothing at all.

### Tablet Device

Wacom CTE-440

### Diagnostic Information

[diag.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13273700/diag.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2023-11-07:
> While the UI does not show 4 aux buttons (it should)

It should not as the tablet does not have 4 aux keys. This tablet only has 2 buttons and a wheel, wheels are not supported by OTD at all currently, see #1367 for that. And please show the resulting full data when pressing the **left** aux button, and the **right** aux button on a **fresh replug** (do NOT touch the wheel in this case).

Simply (using your mouse for everything but pressing the aux buttons):
- tablet > tablet debugger
- enable data recording
- press the left button, release
- press the right button, release
- disable data recording

Upload the file from `%localappdata%\OpenTabletDriver\tablet-data.txt` here.

LostInSwiss, 2023-11-08:
Hello and thank you for getting back to me.

I have uploaded two debug logs for you one for the buttons and one for the scroll bar.

I reset OTD (deleting the json file @  %localappdata%\OpenTabletDriver)
Went to Auxiliary Settings tab there are two buttons 1 & 2
If I set key bindings for these buttons so 1=A and 2=B apply/save
**The buttons do nothing** but the scroll bar (which is not supported) does.
Rolling up crates an A but only once
Rolling down creates a B but only once

The reason I assumed that the scroll bar should function as two buttons is that in debug each scroll up or down crates a single entry in the log as if it where a button that has been pressed, but I can accept that it is unsupported at present. Never the less the auxiliary buttons do not work.
Note the scroll bar behaviour will reset once either of the non functional auxiliary buttons have been pressed.

[tablet-data_buttons.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13295356/tablet-data_buttons.txt)
[tablet-data_scroll.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13295358/tablet-data_scroll.txt)

Thank you

LostInSwiss, 2023-11-08:
in addition i just noticed another strange behaviour relating to the left button which might help with diagnosis. While it will not execute the binding, when pressed twice in notepad the cursor stops blinking until any other key is pressed. this behaviour is repeatable.

LostInSwiss, 2023-11-18:
@jamesbt365 Are there any more details you need?

gonX, 2023-11-19:
The parser (or the parser used) needs to be changed to support the 2 buttons properly, which is not trivial given the parser used for your tablet affects a bunch of other tablets as well, so its mostly a question about someone picking this up and making a PR for it.

Just keep in mind that wheels are not properly implemented yet (see #1367 for that) so they will probably not be fixed before much later. Your tablet has a relative wheel from what I can tell by the tablet data dump you've sent.

The parser used by your tablet expects 4 aux buttons with their state on `report[7]` (8th byte) on (zero-indexed) bit 3, 4, 5, and 6 respectively. Your button reports on bit 6 and 7, the last 2 bits, which is why one of the buttons aren't picked up at all by the parser.

Even odder, the comment specifying wheel reports for the parser used by the CTE-440 indicates that they should be coming on the byte following the aux buttons, but your tablet is not sending this extra byte.

Therefore my conclusion is that the CTE-440 uses the wrong aux parser specified in the configuration we have for it.

LostInSwiss, 2023-11-19:
Thank you for your response.

gonX, 2023-11-24:
I pushed a PR that should fix this: #3029 - an action build is available here: https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/6979158043

As usual, since it is an action build that builds from `master`, its configuration is not compatible with 0.6 versions. That means you will have to create a `userdata` folder next to the provided executables to avoid clobbering your current configurations.

Please let us know if it works.

mjh65, 2024-12-18:
I posted on Discord before finding this issue. My Discord post was a duplicate of this, so can be ignored. I am OK to pick up testing of the candidate PR, but will need to get my fork and build set up first, so could be a few days before I can respond.

Just a quick question. #3029 renames BambooReportParser to BambooV2ReportParser. Would GraphireReportParser be a better name (since the CTE-440 is a marketed as a Graphire tablet) or did I misunderstand the naming convention?

gonX, 2025-01-08:
> I posted on Discord before finding this issue. My Discord post was a duplicate of this, so can be ignored. I am OK to pick up testing of the candidate PR, but will need to get my fork and build set [...]

I missed this. I was planning to rebase this and get this working with 0.6.x. And yes, the parser can be renamed since it only (to our knowledge) fits Graphire tablets at the moment.

mjh65, 2025-01-13:
I've now submitted a candidate fix #3673. Slightly different from #3029, I may have duplicated more files than necessary?
Tested and works OK with my CTE-430 & CTE-440, but not with the other models also affected.
Hope this is useful.

Kuuuube, 2025-09-21:
Fixed in [#4058](https://github.com/OpenTabletDriver/OpenTabletDriver/pull/4058) for 0.6.x.

PR #4058 (2025-09-21): [06x] Backport: Candidate fix for Wacom Graphire AUX buttons.
Backport of #3673 with necessary namespace adjustments applied, and `NearProximity`/`HighConfidence` removed as hover distance was not included in the original PR. The latter concern has been pointed out in the original PR.

Original commit description:

> This fix has been tested with a CTE-430 (no aux buttons!) and a CTE-440 (2 buttons and scroll wheel). Models CTE-630, ET-0405* have also been updated but not tested.
> 
> Mapping of scrollwheel to AuxButtons report has been removed.

Fixes #2992 for 0.6.x - will go in 0.6.6 if we  [... cut]
closed 2025-09-21 by Kuuuube