OpenTabletDriver/OpenTabletDriver #2994: Support Request: my cursor is not able to move the full width of my screen in osu

EWReinE, 2023-11-08:
### What do you need help with?

I was playing osu as I normally do. Shortly after starting (about 3 or 4 maps in) I noticed my cursor felt slower or heavier in some way as I was making many unusual misses. So I paused the game to check the cursor movement and It wouldn't move full width of the screen. I checked by exiting the game and it the cursor was working fine. but in the game, it did not go the full width

### What have you tried?

I tried opening and closing all applications and restarting my computer but it did not fix anything. I also tried changing the size of the area but nothing changed either.

### Diagnostic Information

[diagnostics.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13299829/diagnostics.json)

the diagnostics .json file appears blank for some reason. I've attached it here but on my side it looks blank.

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2023-11-08:
Disable raw input in osu!.

EWReinE, 2023-11-08:
ah that did it. thank you!

closed 2023-11-08 by EWReinE