OpenTabletDriver/OpenTabletDriver #2995: Incorrect display mapping with 3 monitors

SpektacleFR, 2023-11-09:
### Steps to Reproduce

- Install the driver & related software (MacOS Sonoma, M2 Max)
- Have 3 Monitors
- Try to map tablet area to top left monitor with full tablet area

[image]

### Expected Behavior

Correctly allow cursor movement to all areas of the top left monitor, as shown in the Area Mapping window.

### Observed Behavior

Tablet area is mapped only to smaller primary display.  I cannot move it outside of the display unless I offset the area to the top left, outside of the monitors.

### Tablet Device

Wacom CTL-4100

### Diagnostic Information

[OpenTabletDriver Diagnostic Information.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13302737/OpenTabletDriver.Diagnostic.Information.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

gonX, 2023-11-19:
Please try uninstalling all your plugins by removing the plugins folder from the application data directory (location of which described on our wiki [HERE](https://opentabletdriver.net/Wiki/FAQ/General#appdata)) and then restarting the daemon.

Also just to make sure: we don't support monitor hotplug yet - that issue is tracked in #1143. If you've changed your monitor setup since starting OpenTabletDriver you will need to restart it first.

bocchai, 2024-01-15:
Having a similar issue on Arch with KDE Plasma Wayland environment. I'm using 2 monitors (huion kamvas 13 pro and regular monitor) and it seems to map the entire virtual display to my regular monitor only unless I offset it.

jamesbt365, 2025-01-24:
Stale. Please reopen if you still have problems with display mapping.

closed 2025-01-24 by jamesbt365