OpenTabletDriver/OpenTabletDriver #3007: Add support for Huion Kanvas Pro 22 (2019)

ZhangXiangN, 2023-11-16:
### Manufacturer

Huion

### Product Name

GT-2201

### Device Width

589 

### Device Height

344

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

[image]

### Diagnostic Information

[Huion Kanvas 22 Pro (2019).json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13376943/Huion.Kanvas.22.Pro.2019.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-12-26:
This is going to be much easier to do our in our [Discord server](https://discord.com/invite/9bcMaPkVAR), so if you are still interested in getting this tablet supported it would be preferred for you to join here, create a thread in #config-creation and give me a ping.

If you can't join our discord that is okay, we can start here by you providing me with the file generated when you dump strings for your tablet.

tablet > device string reader
9580
110
dump all.

ZhangXiangN, 2023-12-26:
Hi,

Thank you for working on this. I already made a thread in config-creation.
Here you have the link.
https://discord.com/channels/615607687467761684/1174454780001800232

Thank you again for your service.

Kind regards.

PR #3153 (2024-05-29): Add support for Huion Kamvas Pro 22 (2019)
Diagnostic: https://canary.discord.com/channels/615607687467761684/1174454780001800232/1174454780001800232
Strings: https://canary.discord.com/channels/615607687467761684/1174454780001800232/1180885355231187055
Verification: https://canary.discord.com/channels/615607687467761684/1174454780001800232/1189982781040758875

Fixes #3007
closed 2024-05-29 by InfinityGhost