OpenTabletDriver/OpenTabletDriver #3010: Mapping is incorrect on multiple monitor configuration

Tyfaspo, 2023-11-17:
### Steps to Reproduce

1. Connect second monitor.
2. Open OTD
3. Set usable area size to match single display

### Expected Behavior

If area size is matching screen size and coordinates X and Y are set to display 1, usable area must be only at display 1. 
E.g. both screens are 1920*1080, if you set display area to 1920*1080 and use Set to display in context menu to use display 1 you will not be able to move cursor to second screen.
[image]

### Observed Behavior

If area size is matching screen size and coordinates X and Y are set to display 1, usable area is both screens. But if usable area is half of screen width and adding X/2 to X coordinate you can achieve expected behavior. E.g. Width is 1920 / 2 = 960 and X coordinate is 960 + 960 / 2 = 1440 (X = 960 if using 1920*1080 width and height area, then Set to display 1)
[image]

### Tablet Device

Wacom CTL-472

### Diagnostic Information

[diag.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13391944/diag.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

Tyfaspo, 2023-11-17:
[Here youtube video showing unexpected behavior](https://youtu.be/Rv-zl4y1GEY)

jamesbt365, 2023-11-17:
Make sure your desktop environment is not doing any mapping to OTDs virtual tablet as OTD already maps it.

Tyfaspo, 2023-11-17:
TBH I didn't quite know what in my desktop environment is capable of doing such things. I had this problem for a week iirc, today my osu! crashed and with it crashed my kde plasma, I had black screen, I used shortcut to open terminal and ran kstart5 plasmashell as usual to restart desktop(this happened multiple times while mapping was off). But after this I restarted laptop and this problem doesn't exist now. So I guess this issue can be closed?

closed 2023-11-17 by jamesbt365