OpenTabletDriver/OpenTabletDriver #3011: Change detection scheme

X9VoiD, 2023-11-17:
Using just the input/output/feature report length is not specific enough for devices that have all three looking similar.

Instead we should use HID report ID and usage ID.

Something like this in configurations will be enough to cover all possible variations of a tablet and still not be confused by report lengths.

```json
  "Attributes": {
    "HID_REPORT_ID": "0x07",
    "HID_USAGE_PAGE": "0xff0a",
    "HID_USAGE": "0x01",
    "USB_INTERFACE": "1",
  }
```

X9VoiD, 2024-08-25:
#3245 when merged provides the base for finally implementing this.

InfinityGhost, 2024-12-20:
> [#3245](https://github.com/OpenTabletDriver/OpenTabletDriver/pull/3245) when merged provides the base for finally implementing this.

Issue wasn't linked, closing as it is merged.

jamesbt365, 2024-12-20:
This is not complete, the only part that is complete is matching via interface.