OpenTabletDriver/OpenTabletDriver #3012: Support Request: hw1409

quser01, 2023-11-18:
### What do you need help with?

tablet not detected

### What have you tried?

restart the pc
re-install the driver

### Diagnostic Information

[wh1409.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13401619/wh1409.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2023-11-18:
Run TabletDriverCleanup (in the FAQ) then try installing WinUSB (in the installation guide) on various interfaces (probably 0 **OR** 1, installing will cause more issues than it solves) and if it doesn't work, run TabletDriverCleanup again and install on another interface, do not have WinUSB on multiple interfaces at once.

quser01, 2023-11-19:
[wh1409-1.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13402918/wh1409-1.json)

i tried again with and without the tablet driver, but with no results,
forgive me, i can't understand what do you mean with configuration, can you explain it?
thank in advance

jamesbt365, 2023-11-19:
I never mentioned anything to do with configurations, so I'm not sure where you got that from.

But you need to install WinUSB, please refer to what I said previously.

quser01, 2023-11-19:
i've read it again and it was my mistake,
now the tablet is being detected but the pen doesn't move the cursor, the tablet can see the pen

quser01, 2023-11-19:
do i need to install the huion driver?

jamesbt365, 2023-11-19:
> i've read it again and it was my mistake, now the tablet is being detected but the pen doesn't move the cursor, the tablet can see the pen

You have WinUSB on the wrong interface.

> do i need to install the huion driver?

That will break OTD.

quser01, 2023-11-19:
ok, thanks for all the help

closed 2023-11-19 by quser01