OpenTabletDriver/OpenTabletDriver #3017: Packaging should use `postinst` where possible

gonX, 2023-11-19:
Not using this causes any existing drivers to not be unloaded on install (among other things), which may confuse users.

Currently, this is the case on:
- [ ] Arch Linux
- [ ] Gentoo (#2742)

Both Debian and the RPM package seems to use it, so it seems to be the intention that we want to use it everywhere.

jamesbt365, 2024-02-10:
Haven't tested it on gentoo but:
https://github.com/gentoo/guru/blob/master/x11-drivers/OpenTabletDriver-bin/OpenTabletDriver-bin-0.6.3.0.ebuild#L82-L88

This should do the trick?

gonX, 2025-01-24:
> Haven't tested it on gentoo but: https://github.com/gentoo/guru/blob/master/x11-drivers/OpenTabletDriver-bin/OpenTabletDriver-bin-0.6.3.0.ebuild#L82-L88
> 
> This should do the trick?

The text you linked just advises the user to replug. It doesn't unload the drivers.