OpenTabletDriver/OpenTabletDriver #3019: Add support for XP-Pen Artist 22R Pro

RosieGarden66, 2023-11-19:
### Manufacturer

XP-Pen

### Product Name

Artist 22R Pro

### Device Width

476.064mm

### Device Height

267.786mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

The device has 2 scroll wheels and 20 buttons. The pen has 2 buttons.

### Diagnostic Information

[11192023.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13403455/11192023.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-12-26:
This is going to be much easier to do our in our [Discord server](https://discord.com/invite/9bcMaPkVAR), so if you are still interested in getting this tablet supported it would be preferred for you to join here, create a thread in #config-creation and give me a ping.

If you can't join our discord that is okay, just provide the output of `lsusb` here.

Iceball457, 2025-05-03:
Has this tablet been added yet? I wanted to do the configuration myself, but there was no documentation at https://opentabletdriver.net/Wiki/Development/AddingTabletSupport