OpenTabletDriver/OpenTabletDriver #3048: Support drag to pan the scene around in Godot, Unity

chutneyio, 2023-11-26:
### Description

I'm facing an issue when using my XP-Pen (Deco 01 v2) to work with Godot where I can not pan the scene by drag the pen around on MacOS Sonoma. This also happens in Unity.

Other applications like Krita, Aseprite, Gimp... works fine.

I can pan with my pen when using XP-Pen's stock driver. But the stock driver is unstable on my machine so it would be nice if I can use OTD instead. Thanks!

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

chutneyio, 2023-11-26:
It looks like mouse drag event is missing mouse delta x/y information: https://developer.apple.com/documentation/coregraphics/cgeventfield/mouseeventdeltax.

I can pan around after added this before post the drag/move event:

```C#
CGEventSetIntegerValueField(mouseEvent, CGEventField.mouseEventDeltaX, deltaX); 

CGEventSetIntegerValueField(mouseEvent, CGEventField.mouseEventDeltaY, deltaY);
```

I'm not sure if Windows and Linux are also need this or not.

gonX, 2023-11-26:
Is this on 0.6.3.0 or `master` (if so, which commit)? Since I think this is a bug it would help us detect where to look.

PR #3051 (2023-11-26): MacOS: Set mouse delta to move and drag event
This PR fixes the issue where pen drag to pan does not work in some apps which requires mouse delta.
Fixes #3048
closed 2023-11-26 by X9VoiD