OpenTabletDriver/OpenTabletDriver #3057: Parblo A640 V2 not connecting. Potential name mismatch. (Hyprland Arch Linux)

iceeburr, 2023-11-27:
### Steps to Reproduce

1. Connect the tablet to your computer
2. Open the driver
3. Scan for the tablet
4. Tablet not found

### Expected Behavior

The driver should find the tablet connected and work as intended.

### Observed Behavior

The driver can't find the tablet because of a potential mismatch in the name when trying to search for it.
I am no expert in this and I can only speculate, this is why I am reporting this as a bug.
Looking at outputs from `dmesg` we can see the tablet is connecting normally without much happening:
```
[  230.229715] usb 7-1: new full-speed USB device number 3 using xhci_hcd
[  230.359792] usb 7-1: device descriptor read/64, error -71
[  230.596459] usb 7-1: device descriptor read/64, error -71
[  230.826411] usb 7-1: new full-speed USB device number 4 using xhci_hcd
[  230.982642] usb 7-1: New USB device found, idVendor=0483, idProduct=a640, bcdDevice= 0.00
[  230.982648] usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  230.982652] usb 7-1: Product: Parblo A640 V2(F1)
[  230.982655] usb 7-1: Manufacturer:
[  230.999105] input:   Parblo A640 V2(F1) Mouse as /devices/pci0000:00/0000:00:08.3/0000:06:00.3/usb7/7-1/7-1:1.0/0003:0483:A640.000E/input/input33
[  230.999555] input:   Parblo A640 V2(F1) Keyboard as /devices/pci0000:00/0000:00:08.3/0000:06:00.3/usb7/7-1/7-1:1.0/0003:0483:A640.000E/input/input34
[  231.053401] hid-generic 0003:0483:A640.000E: input,hidraw0: USB HID v1.00 Mouse [  Parblo A640 V2(F1)] on usb-0000:06:00.3-1/input0
[  231.058076] input:   Parblo A640 V2(F1) as /devices/pci0000:00/0000:00:08.3/0000:06:00.3/usb7/7-1/7-1:1.1/0003:0483:A640.000F/input/input35
[  231.058465] hid-generic 0003:0483:A640.000F: input,hidraw1: USB HID v1.00 Device [  Parblo A640 V2(F1)] on usb-0000:06:00.3-1/input1
[  231.063310] hid-generic 0003:0483:A640.0010: hiddev96,hidraw2: USB HID v1.00 Device [  Parblo A640 V2(F1)] on usb-0000:06:00.3-1/input2
```
But do you see that? It is `Parblo A640 V2(F1)`, not just `Parblo A640 V2`. My theory is that the driver searches for the name without (F1) leading to it not being found even though it is connected correctly.

Side notes:
On first installation of the opentabletdriver (package from AUR), I could see a message after the installation process saying that "the tablet is connected", I tried and it actually worked. I could indeed use it. I can confirm that I get this message after reinstalling the driver and having my tablet plugged in, but sadly only the first time I installed it I got it to work.
I have actually had a similar issue before which I tried to resolve. It was way more complicated than that because the tablet gave errors when connecting it. I gave up on that a really long time ago, it was with the same tablet under the same operating system (different set-up though).

### Tablet Device

Parblo A640 V2

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13480487/diagnostics.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2023-11-27:
> But do you see that? It is Parblo A640 V2(F1), not just Parblo A640 V2. My theory is that the driver searches for the name without (F1) leading to it not being found even though it is connected correctly.

OTD doesn't even check this before spitting out the tablet. However, this is a variant that requires another identifier.

iceeburr, 2023-11-28:
So what can I do to further resolve the issue?