OpenTabletDriver/OpenTabletDriver #3064: Fedora 39, vk640 , OpenTabletDriver v0.6.3.0

View-my-Git-Lab-krafi, 2023-12-02:
### Description

OpenTabletDriver v0.6.3.0 is not working, fedora 39 , vk640
Fresh install of new fedora,
"Connecting to demon has timed out, verify that Open tablet driver . demon is running"

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

View-my-Git-Lab-krafi, 2023-12-02:
`systemctl --user enable opentabletdriver.service --now`

closed 2023-12-02 by View-my-Git-Lab-krafi

jamesbt365, 2023-12-02:
I insist you read the available templates before opening an issue, I don't see how this is enhancement, given that something is *not working*. Please in the future read the template options and select the appropriate template, doing so would have also shown you a link to the FAQ, which houses the solution to the problem you were having in the first place.