OpenTabletDriver/OpenTabletDriver #3079: Minimize to tray when closing Daemon

PearooXD, 2023-12-09:
### Description

I think it would be nice if the daemon didnt take up space on our taskbar

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

X9VoiD, 2023-12-09:
Use the UI instead. This is impossible to implement in daemon for Windows.

jamesbt365, 2023-12-09:
You can either:
- Start the OTD UX, wait for everything to start working and hard kill the UX in task manager
- run the daemon like this: `Start-Process ./OpenTabletDriver.Daemon.exe -WindowStyle Hidden` (maybe?)
- Use an application or write a script to do this

Both of these methods you will have no tray icon, but as it stands, the UX automatically starts the daemon, meaning you can minimise it and it will go to the tray. There is no need to manually start the daemon unless you are on a system with extremely low specs.

In 0.7, the plan is to make the daemon not start automatically with the UX, and instead provide a service for the driver to start instead. #2772 marks the start of this transition.

closed 2023-12-09 by X9VoiD