OpenTabletDriver/OpenTabletDriver #3104: Huion HS611 has 3 intefaces

kostek001, 2023-12-13:
### Steps to Reproduce

Connect Huion HS611 to computer.

### Expected Behavior

Tablet works normally

### Observed Behavior

OTD will choose one interface randomly.

One interface is for tablet
Second interface is for touchbar
Third idk

### Tablet Device

Huion HS611

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13666715/diagnostics.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2023-12-13:
Please refer to the last issue you opened where i specified that your issue was a duplicate of another. Please do not open any more issues related to this manner, as an issue is tracking multiple identifiers having the same information not being able to be chosen between.

I already explained this is a problem with a dependency we use, and can't be fixed until either we don't use it, or we patch it. Thanks!

kostek001, 2023-12-13:
I understand now. Thanks

closed 2025-12-05 by gonX