OpenTabletDriver/OpenTabletDriver #3106: HoverDistance is not correctly parsed for CTL-672

sorenhoyer, 2023-12-16:
### Steps to Reproduce

1. Open OpenTabletDriver
2. Install Hover Distance Limiter plugin
3. Go to filter settings and set Min/Max Hover Distance to n
4. Click apply/save

### Expected Behavior

When I leave Minimum Hover Distance at 0 and set Maximum Hover Distance to n, I'd expect a valid hover distance to be from the surface and n units up, but not above.

### Observed Behavior

When I check the Tablet Debugger I can see that HoverDistance is 31 from the surface and 5 (or more) millimeters above the surface. 2 cm (ish) above the surface hover distance is 15. It will never report anything outside this range 31-15.

So to be able to have some control (it's not precise as hoverdistance will stay at 31 from 0-5mm (ish)) I have to set Minimum Hover Distance to 29 and Maximum Hover Distance to 31.

I'm not sure if it's relevant to mention that I use "relative mode".

### Tablet Device

Wacom CTL-672

### Diagnostic Information

[error.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13693194/error.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2023-12-16:
This is not a bug, this is completely intended because *this is exactly what the tablet is saying to the computer*, we are actually parsing this correctly.

This is the actual way that Wacom has implemented hover height on this tablet. Some Wacom's report this way, some have hover height ticking the other direction, but the values you see are what the tablet is reporting and are parsed properly.

jamesbt365, 2023-12-16:
We cannot give you more precision than the tablet actually reports to you, so if you want a tablet with more hover height control you will need to unfortunately purchase a "Wacom Pro" tablet (you can find these used usually for pretty cheap depending on location)

> Another observation is that the Raw Tablet Data and Report Rate seems to register even higher than the Tablet Report is able to.

Those reports are out of range reports, they don't contain the information required to move the cursor.

closed 2023-12-16 by jamesbt365

sorenhoyer, 2023-12-16:
I had hoped for some more precision in controlling the hover distance from the surface and a few millimeters up, but I guess the Wacom doesn't support that.

Sorry for assuming it was a bug in your software, and thanks for a quick response.

sorenhoyer, 2023-12-16:
I just tried installing the Wacom driver and while there is unfortunately still some hover distance left when you turn it all the way down, my first impression is that it actually feels like it's a little less than OTD... I'll try testing some more, but so far it seems like there's a slight, yet noticeable difference.