OpenTabletDriver/OpenTabletDriver #3112: enable aarch64 rpm build

space192, 2023-12-18:
### Description

As asahi linux on M-series Mac can now run osu lazer I would love to use my tablet but the script to build on fedora linux assumes x86_64, even though the compilation works ! could you guys enable it ?

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2023-12-18:
Fixed by #3061

space192, 2023-12-18:
nice ! would you recommand that I just jump on the branch or to wait for the official package ?

jamesbt365, 2023-12-18:
Build based off the current 0.6.x branch and just modify the packaging files from x86-64 to arm64

space192, 2023-12-18:
ahah I don't think the PR is ready ahah, I'm still getting mv and cp with x86_64 in the package.sh file in redHat, and exec error with the following output :thinking: 
/usr/bin/otd-gui: line 3: //usr/lib/opentabletdriver/OpenTabletDriver.UX.Gtk: cannot execute binary file: Exec format error

gonX, 2023-12-18:
The PR _should_ make even the RPM builds aarch64 compatible, though I haven't looked entirely at the RPM packages in a while.

Just to confirm, you did these steps?

- Checkout a clean `0.6.x` branch
- Modify `NET_RUNTIME=` line to `NET_RUNTIME="linux-arm64"` in `package.sh`

space192, 2023-12-18:
Yes ! and the NET_RUNTIME env is set with a test done in the modified package.sh file from #3061

space192, 2023-12-18:
so after trying the compilation to tarball rather than .rpm I have everything running ! which means that the only thing with a problem is the rpm creation ahah

gonX, 2025-10-17:
Our build system now supports arm64 since #4179, but the 0.6.x GUI is not guaranteed to work there. If you're testing the `avalonia` branch, it may work better (or not at all). It is not known if the Fedora or Debian build works properly with arm builds yet.

To build for Arm, use `./build.sh --runtime linux-arm64` or similar runtime options.