OpenTabletDriver/OpenTabletDriver #3123: Standardize release asset naming

X9VoiD, 2023-12-22:
Original:
[image]

Renamed:
[image]

Historically, we're doing this rename for `OpenTabletDriver.Web`. We don't have to anymore and we can start using the original filenames once [opentabletdriver.github.io#139](https://github.com/OpenTabletDriver/opentabletdriver.github.io/issues/139) is resolved.

But we can do better.

I'm proposing we use the following rules/convention for naming release assets:

- Use the platform/distro's convention when it exists,
- Else, use `opentabletdriver.<platform>.<version>-<arch>.<ext>`, where:
  - `platform` is `win`, `macos`, or `linux`, this is optional for linux packages.
  - `arch` is `x86_64`, or `arm64`.

gonX, 2024-01-02:
I agree with this, however I would have to see some examples of "use the platform/distro's convention when it exists" - would all of them not fit into the "extension" part of your suggested formatting?

gonX, 2024-01-02:
I got a bit ahead of myself there. The name used for Fedora is a good example of what I was asking about: `opentabletdriver-0.6.4.0-1.x86_64.rpm`

To take inspiration from this, maybe use `opentabletdriver.<platform>.<version>.<arch>.<ext>`?  It replaces `-<arch>` with `.<arch>`. The only downside I see with this is that it might be confusing for people hiding file extensions (making the Windows .exe look like a `.x86_64` file)

I prefer not to include pkgrel (the `-1` in `0.6.4.0-1`), as any packaging updates for these packages would likely have to be fixed by a new minor release anyway, or for the rare occasion that it doesn't need a new release, the old file should be replaced with a fixed version - but if nothing else, I am not completely locked on this idea.

X9VoiD, 2024-01-28:
I agree with dropping the pkgrel. How about `opentabletdriver.<platform>-<version>.<arch>.<ext>`? If we have no clear preference then let's use this one.

related issue comment: https://github.com/OpenTabletDriver/opentabletdriver.github.io/issues/139#issuecomment-1913516934

gonX, 2025-01-26:
Worth noting that Windows and MacOS updater code currently expects the tokens `win-x64` and `osx-x64` respectively to be present in their release names:

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/15aa405cfbb4c34358f7b61b6c668813c050c1c8/OpenTabletDriver.Desktop/Updater/WindowsUpdater.cs#L61
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/15aa405cfbb4c34358f7b61b6c668813c050c1c8/OpenTabletDriver.Desktop/Updater/MacOSUpdater.cs#L36

gonX, 2025-01-26:
After trying to implement this I really don't think the proposed format is great.

It's also not compatible with 0.6.x updater code.

Additionally, `<version>` followed by a period can be confusing to read.

I also suggest capitalizing the project name where possible.

Therefore, I instead propose the following format:
`OpenTabletDriver-<version>_<platform>-<arch>.<ext>`, e.g:
**`OpenTabletDriver-0.6.4.0_win-x64.zip`**
as opposed to OP's proposed format:
`opentabletdriver.win-0.6.4.0.x64.zip`

This also keeps us 0.6.x updater compatible.

If there is opposition to the above format I think we should instead look at what other cross-platform projects do.

For Linux packages, we do likely not want to touch the naming convention too much.

gonX, 2025-08-27:
Fixed with #3762 for 0.6.x and #3786 for avalonia

closed 2025-08-27 by gonX