OpenTabletDriver/OpenTabletDriver #3132: Support for Wacom Cintiq Pro 16 - DTH1620K0

tristengrant, 2023-12-24:
### Manufacturer

Wacom

### Product Name

DTH1620K0

### Pressure Levels

8192

### Diagnostic Information

``
{
  "App Version": "OpenTabletDriver v0.6.3.0+4f8168f42907df4d4c7f539930c25baa72d7886d",
  "Build Date": "2023-12-23",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "6.6.7.1",
    "VersionString": "Unix 6.6.7.1"
  },
  "Environment Variables": {
    "USER": "tristengrant",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":1",
    "WAYLAND_DISPLAY": "wayland-1",
    "PWD": "/home/tristengrant",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/bin:/var/lib/flatpak/exports/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl"
  },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:2c:00.3/usb3/3-3/3-3.4/3-3.4.2/3-3.4.2:1.0/0003:056A:0354.000B/hidraw/hidraw2",
      "Manufacturer": "Wacom Co.,Ltd.      ",
      "ProductName": "Cintiq Pro 16 Touch",
      "SerialNumber": "",
      "FriendlyName": "Cintiq Pro 16 Touch",
      "VendorID": 1386,
      "ProductID": 852,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:2c:00.3/usb3/3-3/3-3.4/3-3.4.1/3-3.4.1:1.0/0003:056A:0350.000C/hidraw/hidraw10",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Cintiq Pro 16",
      "SerialNumber": "7EQ0071001021",
      "FriendlyName": "Cintiq Pro 16",
      "VendorID": 1386,
      "ProductID": 848,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    }
  ],
    {
      "Time": "2023-12-23T18:00:24.7214998-06:00",
      "Group": "DeviceUnauthorizedAccessException",
      "Message": "Not permitted to open HID class device at /dev/hidraw6.",
      "StackTrace": "   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)\n   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)\n   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()\n   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /home/tristengrant/opentabletdriver/src/OpenTabletDriver-0.6.3.0/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /home/tristengrant/opentabletdriver/src/OpenTabletDriver-0.6.3.0/OpenTabletDriver/Extensions.cs:line 12",
      "Level": 0,
      "Notification": false
    },
    {
[... 220 lines cut ...]
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-12-23T18:00:24.7216962-06:00",
      "Group": "DeviceUnauthorizedAccessException",
      "Message": "Not permitted to open HID class device at /dev/hidraw10.",
      "StackTrace": "   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)\n   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)\n   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()\n   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxOutputReportLength()\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_OutputReportLength>b__9_0(HidDevice d) in /home/tristengrant/opentabletdriver/src/OpenTabletDriver-0.6.3.0/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 19\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /home/tristengrant/opentabletdriver/src/OpenTabletDriver-0.6.3.0/OpenTabletDriver/Extensions.cs:line 12",
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-12-23T18:00:24.721703-06:00",
      "Group": "DeviceUnauthorizedAccessException",
      "Message": "Not permitted to open HID class device at /dev/hidraw10.",
      "StackTrace": "   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)\n   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)\n   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()\n   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxFeatureReportLength()\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_FeatureReportLength>b__11_0(HidDevice d) in /home/tristengrant/opentabletdriver/src/OpenTabletDriver-0.6.3.0/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 20\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /home/tristengrant/opentabletdriver/src/OpenTabletDriver-0.6.3.0/OpenTabletDriver/Extensions.cs:line 12",
      "Level": 0,
      "Notification": false
    }
``

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-12-24:
Please remove that mess of "diagnostic" information and instead upload the diagnostics file from help > export diagnostics (which it says to do) or we will be unable to support your tablet.

tristengrant, 2023-12-24:
Hopefully I've fixed it. Sorry about the mess.

jamesbt365, 2023-12-24:
Please restore the original format and upload the entire diagnostics file as an attachment please.

tristengrant, 2023-12-24:
Ok, I've attached the original json file.

jamesbt365, 2023-12-24:
Not really what I meant when I said to restore the original format *and* upload the diagnostic as a file instead, don't bother now. Instead do below.

[70-opentabletdriver.rules.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13760528/70-opentabletdriver.rules.txt)
place this file as `/etc/udev/rules.d/70-opentabletdriver.rules` then do `sudo udevadm control --reload-rules && sudo udevadm trigger` then export another diagnostic and upload it here.

tristengrant, 2023-12-24:
[Wacom Cintiq Pro 16 DTH1620K0 - New.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13760536/Wacom.Cintiq.Pro.16.DTH1620K0.-.New.json)

Looks the same as the other to be, but I could be wrong. Would reinstalling OTD help?