OpenTabletDriver/OpenTabletDriver #3135: Wrong Tablet Area Bug

mycustomusername, 2023-12-25:
### Steps to Reproduce

1. Open driver
2. Check if there is a bug
I dont deal with this bug every time I open the driver, but it happened a lot of times before

### Expected Behavior

Almost Full Tablet Area

### Observed Behavior

Wrong Tablet Area. like 2/3 of area that I want to work with

https://github.com/OpenTabletDriver/OpenTabletDriver/assets/100908480/34fcb58d-36ac-41a8-afdf-a8efc7ca0cdb

### Tablet Device

Wacom CTH-480

### Diagnostic Information

N/A

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2023-12-25:
Diagnostic information is extremely useful in circumstances like this, please provide it.

mycustomusername, 2023-12-25:
[bug.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13766145/bug.json)

my desktop resolution is 1680x1050 and my osu game is running on 1280x800
i figured out that if opentabletdriver load while osu is an active window, it changes area size to fit in-game resolution. 

driver looks like this if i alt tab then
[image]

and if i drag display area it turns to this. expected to have area that i set
[image]

jamesbt365, 2023-12-25:
OpenTabletDriver is unaware of display resolution changes, see #1143

closed 2023-12-25 by jamesbt365