OpenTabletDriver/OpenTabletDriver #3159: Add support for XP-Pen-Artist24

Genthus, 2023-12-30:
### Manufacturer

Xp-pen

### Product Name

Artist 24

### Device Width

590mm

### Device Height

361mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

Additional specifications:
'''
Product Name | Artist 24
-- | --
Product Model | CD240Q
Dimensions | 590 × 361 × 34mm
Active Area | 526.85 × 296.35mm
Display Resolution | 2560 × 1440
Color Gamut (typical) | 90% NTSC, 94% Adobe RGB, 127% sRGB
Mechanical Buttons | 5
Stylus | Battery-free
Tilt | 60°
Pressure Levels | 8192
Contrast (typical) | 1000:1
Resolution | 5080LPI
Report Rate | 220RPS (max)
Reading Height | 10mm
Accuracy | ±0.5mm (center), ±1.5mm (corner)
Ports Included | 1 x USB-C, 1x HDMI, 1 x DC-IN, 1 x Headphone
Viewing Angle | 178°
Adjustable Stand | 16°-90°
Brightness | 250 cd/m2
VESA Mount | 100 × 100mm
Power Input | DC 12V/3A
Compatibility | Windows 7 (or later), Mac OS X 10.10 (or later), Chrome OS 88 (or later), Linux(This Product now fully Supports macOS Monterey and Windows 11.)
Certifications | FCC/CE/EAC/NOM/PSE/WEEE/RCM/KC/RoHs/BIS
'''

### Diagnostic Information

```
{
  "App Version": "OpenTabletDriver v0.6.4.0+7b98ae8115e65adb19804deaf47ce73bc99b5233",
  "Build Date": "12/30/2023",
  "Operating System": {
    "Name": "Arch Linux",
    "Version": "Unknown",
    "Attributes": {
      "PRETTY_NAME": "Arch Linux",
      "ID": "arch",
      "BUILD_ID": "rolling",
      "KERNEL_VERSION": "6.6.8-arch1-1"
    }
  },
[... 2156 lines cut ...]
    },
    {
      "Time": "2023-12-29T18:55:43.6673512-06:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-12-29T18:56:18.0792161-06:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-12-30:
I'm currently taking configuration requests on discord in our #config-creation forum, so I'll get around to handling this faster over in [our Discord](https://discord.com/invite/9bcMaPkVAR). As creating a configuration also requires more information from you later and you testing a configuration discord is a more optimal platform for this.

If you don't have a Discord account or do not want to create one, that is fine and I'll get back to you at some point in the near future.

jamesbt365, 2025-01-25:
Fixed by #3744.

PR #3744 (2025-01-25): Add support for XP-Pen Artist 24
resolves #3159
commit 9b16fe91 (2025-01-25): Add files via upload

resolves #3159
closed 2025-01-25 by jamesbt365