OpenTabletDriver/OpenTabletDriver #3167: Simultaneous Auxiliary Bidding does not work

MotaOcimar, 2024-01-04:
### Steps to Reproduce

Set `Auxiliary Bidding 1` to `LeftControl` and set `Auxiliary Bidding 2` to `C`.
Save, select a text and click the Auxiliary Buttons 1 and 2 simultaneously.

### Expected Behavior

A `control+C` command must be sent.

### Observed Behavior

Only the first Auxiliary Button clicked sends a bidding (only `control`)

### Tablet Device

Gaomon S620

### Diagnostic Information

I can't past. Github says `Tere was an error creating your Issue: body is too long (maximum is 65536 characters).`

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-01-04:
Just tested this and it works fine for me. I can press all the buttons at once with no trouble. Ensure your tablet is registering buttons correctly in the tablet > tablet debugger.

And you can also upload the json file directly, which I must prefer people doing over pasting.

MotaOcimar, 2024-01-04:
Thank you for the ideas!

I went to `tablet > tablet debugger` and found that only one key is being registered at a time.
My tablet is a Gaomon S620. Is there something we can do?

And here is the [Diagnostic Information.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13834344/Diagnostic.Information.json)

jamesbt365, 2024-01-04:
You can try running TabletDriverCleanup from the [FAQ](https://opentabletdriver.net/Wiki/FAQ/Windows#driver-cleanup), the log messages from your diagnostic information show that some remnants of the Gaomon driver exist.

If running it and uninstalling everything it asks doesn't rectify your issue, it would lie with the tablet as the parser the tablet uses in OTD support multiple button presses at once.

The tablet debugger's raw data shows what the tablet is reporting so if the data doesn't allow for both to be pressed unfortunately there isn't anything that can be done in the driver.

MotaOcimar, 2024-01-04:
I tried the TabletDriverCleanup without success.

> The tablet debugger's raw data shows what the tablet is reporting so if the data doesn't allow for both to be pressed unfortunately there isn't anything that can be done in the driver.

Unfortunately the raw data is the same for one or two keys being pressed 😞
So, I think it just isn't possible for Gaomon S620 Tablets.

closed 2024-01-04 by MotaOcimar