OpenTabletDriver/OpenTabletDriver #3170: Support Request: Disable update check/can't be blocked with firewall

HaveDex, 2024-01-05:
### What do you need help with?

I don't want the program to check if there's an update available when I launch it. Blocking it (any of the executables) via the firewall does nothing, it still tells me there's an update. Unplugging my ethernet prior will have it check for a bit before giving up, so it's not like it has a cache file that says "there's a new version". It somehow access the internet despite being blocked. I don't want that.

### What have you tried?

Tried blocking via Windows Firewall, which did nothing.

### Diagnostic Information

/

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [ ] I am on the latest version of OpenTabletDriver.

HaveDex, 2024-01-05:
Daemon.exe is the one which accessing the internet, 140.82.121.6 and 140.82.121.5 and so on, Github's IPs. I don't want it to access the internet at all.

Kuuuube, 2024-01-05:
Set up the firewall correctly. Here is a guide https://www.wikihow.com/Block-a-Program-with-Windows-Firewall

Method 1 listed here will work. I have tested it. 

If you would like to bring OTD entirely offline instead of just the updater you should also repeat these steps for `OpenTabletDriver.UX.Wpf.exe`

gonX, 2025-08-27:
Being able to ignore updates is currently not planned as there is a workaround (using the firewall). Please open a new feature request for disabling updates.

gonX, 2026-11-12:
Fixed in #4270 - use `--skipupdate` for the command arguments on the GUI.

closed 2026-11-12 by gonX