OpenTabletDriver/OpenTabletDriver #3171: Tablet stops working after resuming from suspend

khoidauminh, 2024-01-06:
### Steps to Reproduce

1. Start OpenTabletDriver service, plug tablet and confirm that it's working
2. Suspend the system
3. Resume

### Expected Behavior

Tablet should work as normal.

### Observed Behavior

Tablet stops working. Cursor doesn't move when using the stylus. Buttons do not work.
Tablet LED lights up when stylus comes into contact but nothing happens on the screen.
OpenTabletDriver GUI tool still detects the tablet, but it still does not work even after changing and applying settings.

### Additional behavior:

Running `otd detect` in the terminal yields:
```
Thg 1 05 20:13:11 kpc otd-daemon[102978]: [Device:Debug]        A device stream was disposed.
Thg 1 05 20:13:11 kpc otd-daemon[102978]: [IPC:Debug]        Disconnected from a client.
```
After searching for tablets

Manually running Detect tablet from the GUI fixes the issue 
Unplugging the tablet and replugging fixes the issue

### Tablet Device

Gaomon 1060Pro

### Diagnostic Information

[diag.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13842425/diag.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

gonX, 2024-10-19:
Seems to me like #2236 needs a backport. I've added the label to that PR.

gonX, 2024-10-19:
Apparently, that one has been backported for a while, not sure why your diag is not showing a sleep at all then.

If you suspended for less than a couple of seconds (8 to be exact), the suspend won't be detected.

khoidauminh, 2024-10-20:
Been a long while and I don't remember much, but I'll get back to this when I get time

khoidauminh, 2024-10-20:
Here's my system logs when I suspend

```
Thg 10 20 08:44:59 kpc systemd-logind[694]: The system will suspend now!
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.0537] manager: sleep: sleep requested (sleeping: no  enabled: yes)
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.0538] device (eno1): state change: unavailable -> unmanaged (reason 'unmanaged-sleeping', managed-type: 'full')
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.0624] device (p2p-dev-wlo1): state change: disconnected -> unmanaged (reason 'unmanaged-sleeping', managed-type: 'full')
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.0625] manager: NetworkManager state is now ASLEEP
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.0627] device (wlo1): state change: activated -> deactivating (reason 'sleeping', managed-type: 'full')
Thg 10 20 08:44:59 kpc systemd[1]: Starting Network Manager Script Dispatcher Service...
Thg 10 20 08:44:59 kpc systemd[1]: Started Network Manager Script Dispatcher Service.
Thg 10 20 08:44:59 kpc kernel: wlo1: deauthenticating from 28:77:77:b0:23:81 by local choice (Reason: 3=DEAUTH_LEAVING)
Thg 10 20 08:44:59 kpc wpa_supplicant[783]: nl80211: send_event_marker failed: Source based routing not supported
Thg 10 20 08:44:59 kpc wpa_supplicant[783]: wlo1: CTRL-EVENT-DISCONNECTED bssid=28:77:77:b0:23:81 reason=3 locally_generated=1
Thg 10 20 08:44:59 kpc wpa_supplicant[783]: wlo1: CTRL-EVENT-DSCP-POLICY clear_all
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.2857] device (wlo1): supplicant interface state: completed -> disconnected
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.2857] device (wlo1): state change: deactivating -> disconnected (reason 'sleeping', managed-type: 'full')
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Withdrawing address record for fe80::2ccb:a07b:537d:3b96 on wlo1.
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.2859] dhcp4 (wlo1): canceled DHCP transaction
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Leaving mDNS multicast group on interface wlo1.IPv6 with address fe80::2ccb:a07b:537d:3b96.
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.2859] dhcp4 (wlo1): activation: beginning transaction (timeout in 45 seconds)
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Interface wlo1.IPv6 no longer relevant for mDNS.
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.2859] dhcp4 (wlo1): state changed no lease
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Interface wlo1.IPv4 no longer relevant for mDNS.
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Leaving mDNS multicast group on interface wlo1.IPv4 with address 192.168.2.13.
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.3070] device (wlo1): set-hw-addr: set MAC address to 9A:51:B5:75:EE:F9 (scanning)
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Withdrawing address record for 192.168.2.13 on wlo1.
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Joining mDNS multicast group on interface wlo1.IPv4 with address 192.168.2.13.
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: New relevant interface wlo1.IPv4 for mDNS.
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Registering new address record for 192.168.2.13 on wlo1.IPv4.
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Withdrawing address record for 192.168.2.13 on wlo1.
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Leaving mDNS multicast group on interface wlo1.IPv4 with address 192.168.2.13.
Thg 10 20 08:44:59 kpc avahi-daemon[2144]: Interface wlo1.IPv4 no longer relevant for mDNS.
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.4162] device (wlo1): state change: disconnected -> unmanaged (reason 'unmanaged-sleeping', managed-type: 'full')
Thg 10 20 08:44:59 kpc gnome-shell[1462]: Cursor update failed: drmModeAtomicCommit: Đối số không hợp lệ
Thg 10 20 08:44:59 kpc NetworkManager[686]: <info>  [1729388699.5694] device (wlo1): set-hw-addr: reset MAC address to F0:57:A6:FA:98:FD (unmanage)
Thg 10 20 08:44:59 kpc wpa_supplicant[783]: p2p-dev-wlo1: CTRL-EVENT-DSCP-POLICY clear_all
Thg 10 20 08:44:59 kpc wpa_supplicant[783]: p2p-dev-wlo1: CTRL-EVENT-DSCP-POLICY clear_all
Thg 10 20 08:44:59 kpc wpa_supplicant[783]: nl80211: deinit ifname=p2p-dev-wlo1 disabled_11b_rates=0
Thg 10 20 08:44:59 kpc wpa_supplicant[783]: wlo1: CTRL-EVENT-DSCP-POLICY clear_all
Thg 10 20 08:44:59 kpc wpa_supplicant[783]: wlo1: CTRL-EVENT-DSCP-POLICY clear_all
Thg 10 20 08:44:59 kpc wpa_supplicant[783]: nl80211: deinit ifname=wlo1 disabled_11b_rates=0
Thg 10 20 08:45:00 kpc systemd[1]: Reached target Sleep.
Thg 10 20 08:45:00 kpc systemd[1]: Starting NVIDIA system suspend actions...
Thg 10 20 08:45:00 kpc suspend[5510]: nvidia-suspend.service
Thg 10 20 08:45:00 kpc logger[5510]: <13>Oct 20 08:45:00 suspend: nvidia-suspend.service
Thg 10 20 08:45:00 kpc bluetoothd[687]: Endpoint unregistered: sender=:1.177 path=/MediaEndpoint/A2DPSource/ldac
Thg 10 20 08:45:00 kpc bluetoothd[687]: Endpoint unregistered: sender=:1.177 path=/MediaEndpoint/A2DPSink/aptx_hd
Thg 10 20 08:45:00 kpc bluetoothd[687]: Endpoint unregistered: sender=:1.177 path=/MediaEndpoint/A2DPSource/aptx_hd
Thg 10 20 08:45:00 kpc bluetoothd[687]: Endpoint unregistered: sender=:1.177 path=/MediaEndpoint/A2DPSink/aptx
Thg 10 20 08:45:00 kpc bluetoothd[687]: Endpoint unregistered: sender=:1.177 path=/MediaEndpoint/A2DPSource/aptx
[... cut ...]
Thg 10 20 08:46:03 kpc otd-daemon[2005]: [Detect:Info]        No tablets were detected.
Thg 10 20 08:46:03 kpc gdm[786]: GLib: Source ID 83 was not found when attempting to remove it
Thg 10 20 08:46:03 kpc systemd[1]: systemd-suspend.service: Deactivated successfully.
Thg 10 20 08:46:03 kpc systemd[1]: Finished System Suspend.
Thg 10 20 08:46:03 kpc systemd[1]: Stopped target Sleep.
Thg 10 20 08:46:03 kpc systemd[1]: Reached target Suspend.
Thg 10 20 08:46:03 kpc systemd-logind[694]: Operation 'suspend' finished.
```

khoidauminh, 2024-10-20:
New diagnostics file

[diag.json](https://github.com/user-attachments/files/17448288/diag.json)

Skaifer, 2024-12-16:
I have the same problem on Bazzite with Gaomon M10K Pro. Now I'm temporary on PikaOS, but I want to switch back, because of Steam related issues. Now, the tablet and driver are working after waking up from suspend on Pika. Why is that? Is there anything I can share from Pika to help fix this problem, whilst I'm on the boat? 
Kinda sick of unpluging and pluging my tablet every time after sleepy-wakey for more than a year straight.

Fernande-1, 2025-03-26:
Hello everyone! I use "Fedora" as the main system and "XP-Pen Star G640" as my only graphic tablet. 😎 I have the same problem, but here's the interesting thing: the tablet starts working if I replug it. 🤔 I asked ChatGPT to make a script that replugs the device programmatically, and it worked! If you're reading this facing the same problem, I've got a little guide for you. I wouldn't say I'm good at Linux or programming, but it works for me, so I think I should share it. 🤠 First of all, how the script works.
**When the system wakes up from sleep mode, this script automatically runs**. It:

- Searches for all connected USB devices.
- Finds the correct device by matching both the Vendor ID and Product ID—those are information that is unique for every USB device on your PC.
- Identifies the device's USB path (which can change depending on the USB port used; changing the port will not break the script).
- Temporarily unbinds the device from the USB driver.
- Rebinds it to make the system recognize it again, just like a manual replugging.

Now, here's the guide itself:

1. In the terminal, run `lsusb`. You will see the list of devices connected to your PC. Find the string with your tablet name. In my case: `Bus 001 Device 004: ID 28bd:0094 XP-Pen STAR G640`. The information we need is `28bd:0094`—those are Vendor ID and Product ID. **It will be different for your tablet**; remember it or write down somewhere. ✍️
2. Create the script using `sudo nano /lib/systemd/system-sleep/replug-tablet.sh`.
3. When you are in nano, paste this script:
```
#!/bin/bash

case "$1" in
    post)
	echo "Searching for XP-PEN Star G640..."

        # Find the correct USB device path dynamically using both Vendor ID (28bd) and Product ID (0094)
        DEVICE_PATH=$(grep -l "28bd" /sys/bus/usb/devices/*/idVendor | while read VENDOR_PATH; do
            DEVICE_DIR=$(dirname "$VENDOR_PATH")
            if grep -q "0094" "$DEVICE_DIR/idProduct"; then
                basename "$DEVICE_DIR"
                break
            fi
	done)

	if [ -n "$DEVICE_PATH" ]; then
            echo "Tablet found at: $DEVICE_PATH"
            echo "$DEVICE_PATH" | sudo tee /sys/bus/usb/drivers/usb/unbind
            sleep 1
            echo "$DEVICE_PATH" | sudo tee /sys/bus/usb/drivers/usb/bind
            echo "Tablet successfully reconnected!"
        else
            echo "Tablet not found! Maybe it's unplugged?"
        fi
	;;
esac
```
4. Replace` XP-PEN Star G640`, `28bd` and `0094` with the information of your tablet.
5. Press `Ctrl+S` to save the script.
6. Press `Ctrl+X` to exit nano.
7. When you leave nano and when you are in the usual terminal, make the script executable using `sudo chmod +x /lib/systemd/system-sleep/replug-tablet.sh`.
8. Test the script manually using `/lib/systemd/system-sleep/replug-tablet.sh post`. If your tablet disconnects and reconnects, the script works! 🎉🎉🎉
9. Try to use it in a real situation: suspend your PC using `sudo systemctl suspend`  and wake it up by pressing any keyboard button or clicking any mouse button. If your tablet really works, I was happy to help you! 🙏 If it doesn't, my first recommendation is to ask ChatGPT or Grok how to fix the script for your Linux distribution and share information here with other users. (If you understand something in Linux, you can try to create your own decision. Feel free to share it here too—we will be happy to try using it and help you with an improvement of the script. ❤️‍🔥❤️‍🔥)
Thank you for reading!

Prayag2, 2025-04-27:
The tablet also starts working upon restarting the systemd service using `systemctl --user restart opentabletdriver.service`. So you can create a resume script that runs this command on every wakeup!

MathisP75, 2026-02-02:
Same issue here. I get the following in the systemd logs:
```
Feb 01 20:06:18 LEGION-Mathis otd-daemon[535113]: [Settings:Info]        Driver is enabled.
Feb 01 20:06:18 LEGION-Mathis otd-daemon[535113]: [NullReferenceException:Error]        Object reference not set to an instance of an object.
Feb 01 20:06:18 LEGION-Mathis otd-daemon[535113]:    at OpenTabletDriver.Devices.DeviceReader`1.Main() in /home/mathis-kde/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.6.2/OpenTabletDriver/Devices/DeviceReader.cs:line 109
Feb 01 20:06:53 LEGION-Mathis otd-daemon[535113]: [IPC:Debug]        Disconnected from a client.
```

philg-dev, 2026-08-24:
I'm not entirely sure if it's the same issue as here, but since nobody here posted a full stack trace from the crash handler popup yet, I'm still posting it, hoping that it might provide some useful insights for the devs.

I just got back home and after suspend I was greeted with, what seemed to be a never ending loop of Error popups with stacktraces from OpenTabletDriver.

```log
System.NullReferenceException: Object reference not set to an instance of an object.
   at Eto.GtkSharp.Forms.Controls.GridHandler`2.get_SelectedRows()
   at Eto.GtkSharp.Forms.Controls.GridHandler`2.UpdateModel()
   at Eto.GtkSharp.Forms.Controls.GridViewHandler.CollectionHandler.AddRange(IEnumerable`1 items)
   at Eto.CollectionChangedHandler`2.CollectionChanged(Object sender, NotifyCollectionChangedEventArgs e)
   at OpenTabletDriver.UX.Tools.LogDataStore.OnCollectionChanged(NotifyCollectionChangedAction action, Object obj) in ${HOME}/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.7/OpenTabletDriver.UX/Tools/LogDataStore.cs:line 68
   at OpenTabletDriver.UX.Tools.LogDataStore.Add(LogMessage message) in ${HOME}/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.7/OpenTabletDriver.UX/Tools/LogDataStore.cs:line 54
   at OpenTabletDriver.UX.Controls.LogView.<>c__DisplayClass4_0.<AddMessage>b__0() in ${HOME}/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.7/OpenTabletDriver.UX/Controls/LogView.cs:line 137
   at GLib.Timeout.TimeoutProxy.Handler()
```

Note: I redacted my user directory by replacing it with `${HOME}`

My usage scenario was probably quite exotic:
- I plugged in my tablet earlier today and it worked perfectly fine
- I even used the OTD GUI to change my tablet settings, since I wanted a different active area for a moment
- I finished my task with the tablet and unplugged it again, while OTD GUI was still open in the background
- I left the house for a couple of hours and suspended my laptop as usual
- came back to a plethora of errors as provided above (there's probably many of them, as they have to do with some sort of `Timeout`, which might have accumulated quite a bit over several hours of being away?

I killed the OTD process via GNOME System Monitor to break out of the seemingly infinite error loop. (While I was at it, I also killed the Daemon process, which was restarted automatically by systemd.)

My system:
- Arch Linux Kernel 7.1.8-arch1-3
- GNOME 50.4 (Wayland, gdm)
- OTD v0.6.7 (installed from the [AUR package `opentabletdriver 0.6.7-2`](https://aur.archlinux.org/packages/opentabletdriver)
- Tablet: Wacom Bamboo Pen CTL-470

Thanks for your time and efforts!

DDunda, 2026-09-16:
Hi, just adding that I have encountered this issue and patched it with a systemd service after some trial and error. Place the following into a file called `restart-otd@.service` in `/etc/systemd/system/`:
```ini
[Unit]
Description=Restarts OpenTabletDriver after suspension
After=suspend.target

[Service]
Type=simple
ExecStart=systemctl --user --machine=%i@ restart opentabletdriver.service

[Install]
WantedBy=suspend.target
```
Then run the following in the terminal, and OTD should restart after suspension:
```
sudo systemctl daemon-reload
sudo systemctl enable restart-otd@$(whoami)
```
Thanks to Daiel Garajau for writing up how to run a user service that reacts to suspension:
https://garajau.com.br/2022/08/systemd-suspend-user-level