OpenTabletDriver/OpenTabletDriver #3176: I can't click when I plug my tablet

QXR9, 2024-01-07:
### Steps to Reproduce

1 - Download and install OpenTabletDriver.
2 - Plug 10moons T503 in with USB
3 - Click plug in and wait for 5 seconds.

### Expected Behavior

Normally my tablet shouldn't bug and let me click.

### Observed Behavior

I can't click even if I unplug the tablet. My computer completely bugging and I have to restart.

### Tablet Device

10moons T503

### Diagnostic Information

[Diagnostic.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13853979/Diagnostic.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-01-07:
Assuming your tablet detects under 10moons 1060N, as thats the only 10moons we support and a lot of tablets use that digitizer, the config is broken, there is already issues and pull requests open in attempts to fix it, but as it stands this is a terrible tablet and you currently can't use it with OTD.

Use the OEM driver if it has one, otherwise you'll have no driver at all.

closed 2024-01-07 by jamesbt365

QXR9, 2024-01-07:
It have but It doesn't work for linux. It is the real problem. I think buying a new tablet which open driver supports going to solve this problem. Thanks!