OpenTabletDriver/OpenTabletDriver #3180: Support Request: OTD doesn't recognize display tablet as secondary display

lupelagic, 2024-01-16:
### What do you need help with?

I am currently trying to use a Wacom One display tablet on Fedora Linux 39, and OTD just doesn't recognize my tablet as a secondary display. It shows the screen area of both the laptop and tablet display, but it seems like it just merges the two into the laptop display, making it so that I can't really use my screen tablet as intended. 
[image]
(laptop display is left, tablet display is right)
When the settings are like this, it basically assumes the whole display area is just my laptop display, and so the input will cover about half of my laptop screen. 

### What have you tried?

I've fixed this issue before but quite frankly I don't remember how. I've tried restarting the program, unplugging and replugging in the tablet, restarting the laptop, but no dice.

### Diagnostic Information

[diagOTD.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13949073/diagOTD.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-01-16:
KDE Wayland?

lupelagic, 2024-01-16:
> KDE Wayland?

omfg of course it's wayland. yep. switched to x11 and it works just fine. thank you!

jamesbt365, 2024-01-16:
Yep looking into the diagnostic you are on wayland, this problem is external to us and requires KDE to allow mapping to the entire virtual desktop (or allow for disabling mapping), because your displays are different resolutions, its effectively impossible to map in KDE wayland properly at the moment, so either use X11 or another desktop environment.

closed 2024-01-16 by jamesbt365