OpenTabletDriver/OpenTabletDriver #3182: Add support for Huion H1061P

cssuperman, 2024-01-17:
### Manufacturer

Huion

### Product Name

H1061P

### Device Width

266.7

### Device Height

166.7

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Diagnostic Information

"HID Devices": [
    {
      "DevicePath": "\\\\?\\usb#vid_256c&pid_0068&mi_00#6&6465fb8&0&0000#{dee824ef-729b-4a0e-9c14-b7117d33a817}",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_H1061P",
      "SerialNumber": "C6T12SH00001",
      "FriendlyName": "Huion Tablet_H1061P",
      "VendorID": 9580,
      "ProductID": 104,
      "InputReportLength": 64,
      "OutputReportLength": 64,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": null
    },
    {
      "DevicePath": "\\\\?\\usb#vid_256c&pid_0068&mi_01#6&6465fb8&0&0001#{dee824ef-729b-4a0e-9c14-b7117d33a817}",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_H1061P",
      "SerialNumber": "C6T12SH00001",
      "FriendlyName": "Huion Tablet_H1061P",
      "VendorID": 9580,
      "ProductID": 104,
      "InputReportLength": 16,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": null
    },
    {
      "DevicePath": "\\\\?\\usb#vid_256c&pid_0068&mi_02#6&6465fb8&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817}",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_H1061P",
      "SerialNumber": "C6T12SH00001",
      "FriendlyName": "Huion Tablet_H1061P",
      "VendorID": 9580,
      "ProductID": 104,
      "InputReportLength": 64,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": null
    }

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-01-17:
You have the huion driver installed, this prevents us from supporting the tablet.

cssuperman, 2024-01-17:
> You have the huion driver installed, this prevents us from supporting the tablet.

[huion H1061P.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13964037/huion.H1061P.json)

Oh sorry,I runed tabletdrivercleanup and get this data for you.

jamesbt365, 2024-01-17:
tablet > device string reader
9580
104
dump all, save that file and send it here.

cssuperman, 2024-01-17:
> tablet > device string reader 9580 104 dump all, save that file and send it here.

[Detect:Info]	No tablets were detected.

I click tablet > device string reader,get nothing

jamesbt365, 2024-01-17:
If by that you mean no menu pops up, please follow the warning that you get when you open the driver, opening it as Administrator is not supported.

cssuperman, 2024-01-17:
> If by that you mean no menu pops up, please follow the warning that you get when you open the driver, opening it as Administrator is not supported.

I didn't ran as Administrator

jamesbt365, 2024-01-17:
Its evident that you are infact running the driver as Administrator because OTD is warning you that it is elevated and menus aren't opening (which is a problem with a dependency of OTD when its elevated), so please check you have UAC enabled and that you aren't using the account called Administrator.

Please note that we do not support custom Windows installations, as they do modifications like this that prevent OTD and other applications from not working, causing technical debt we don't have the resources to handle.

So until you can get a string reading, which will require you to not run OTD as admin, we cannot proceed further.

cssuperman, 2024-01-17:
> Its evident that you are infact running the driver as Administrator because OTD is warning you that it is elevated and menus aren't opening (which is a problem with a dependency of OTD when its [...]

[223.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13968726/223.txt)

I changed UAC Level and got this.

jamesbt365, 2024-01-17:
## Incomplete configuration

[H1061P.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13969243/H1061P.json)

You can try this configuration, this will likely detect, but it will *not* work in full, I require extra information that you must provide me.

First create a folder called `Configurations` next to the driver executables (examples include `OpenTabletDriver.UX.Wpf.exe` & `OpenTabletDriver.Daemon.exe`)

Place the file I have linked into the newly created folder called `Configurations.`

## Specification verification
Once you have done that replug your tablet or restart the driver, then navigate to `Tablets > Tablet Debugger`.

The information I need you to provide here is:

- The max x and y coordinates when you move your pen to the bottom right of the tablet
- Pen pressure when you press the pen tip all the way in.
- Check if the pen buttons go to `true` in the debugger when you press them.

## Auxiliary key data recording
I also need verification on the positions and functionality of the auxiliary keys.

This part may be a little tricky.

The H1061P has 3 group buttons, 8 barebones auxiliary keys and 1 roller.

The roller is not supportable at the current moment, and the group buttons may act like regular Auxiliary keys, but I need that information for verification purposes.

Press `Enable Data Recording` with your mouse (please don't use your tablet pen for this process)

Press the 3 group buttons from top to bottom, then press the 8 auxiliary buttons from top to bottom.

Disable data recording and upload the file from `%localappdata%\OpenTabletDriver\tablet-data.txt`

cssuperman, 2024-01-18:
> ## Incomplete configuration
> [H1061P.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13969243/H1061P.json)
>
> You can try this configuration, this will likely detect, but it will _not_ [...]

[tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13977962/tablet-data.txt)

max x and y coordinates：53340,33340
Pen pressure：8191
pen buttons go to true：true

jamesbt365, 2024-01-18:
did you go through all the buttons? I only see 7 unique auxiliary states?

Could you do a separate recording for just the group buttons and upload that here?

cssuperman, 2024-01-19:
> did you go through all the buttons? I only see 7 unique auxiliary states?
> 
> Could you do a separate recording for just the group buttons (not the regular aux) and upload that here?

I found the 19th line is incomplete in last tablet-data file , maybe there is a bug about data recording.
Here is 2 types buttons data.
[GroupButton_tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13990677/GroupButton_tablet-data.txt)

[A_Button_tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13990676/A_Button_tablet-data.txt)

cogsfx, 2024-03-28:
Sorry for the delay, adding to this my H1061P is not detected at all in linux. I have the linux driver that I can use from Huion which works on x11, but on wayland that is just not working, I mean its working but it's mapped on a single monitor and doesn't detect my 2 monitors.

BreakingLead, 2024-09-16:
So are there any progress?

PR #3624 (2024-12-22): Add support for Huion H1061P
Fixes #3182, verification is within that issue alongside how the group keys report data.
closed 2024-12-22 by gonX