OpenTabletDriver/OpenTabletDriver #3183: Add support for Huion HST640

FloatingMind12, 2024-01-17:
### Manufacturer

Huion

### Product Name

HST640

### Device Width

160mm

### Device Height

102mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

The tablet seems to be a rip-off version of Huion HS64 as they are entirely similar

### Diagnostic Information

[hst640.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13969497/hst640.txt)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-01-17:
tablet > device string reader
9580
110
dump all, upload that file here.

FloatingMind12, 2024-01-17:
Which number goes where?
There are DeviceID, ProductID and String Index

jamesbt365, 2024-01-17:
The numbers are in order of which you need to place them and you don't need an index because you are dumping *all* indexes.

FloatingMind12, 2024-01-17:
But nothing happens when I press Dump All

jamesbt365, 2024-01-17:
The dumping processes takes a while on some tablets. Give it a few minutes, if it still doesn't prompt you to save a file, send a request for index 200 and 201.

FloatingMind12, 2024-01-17:
Here is the file
[dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13969812/dump.txt)

jamesbt365, 2024-01-18:
[HS64.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13970205/HS64.json)
place this in `~/.local/share/OpenTabletDriver/Configurations` (you need to create this directory), then replug your tablet and go into the tablet > tablet debugger.

Get the max x and y when moving your pen to the bottom right of the debugger, get the max pressure as well and check if the pen and tablet buttons go to true when you press them.

FloatingMind12, 2024-01-18:
Max X: 32000
Max Y: 20400
Max Pressure: 8191
Pen Buttons work fine
About tablet buttons: there's nothing written about them in the tablet report section but some text that I couldn't catch flicker there when I press them

jamesbt365, 2024-01-18:
TIP: If you dont have your pen on the tablet you can see the aux reports better.

FloatingMind12, 2024-01-18:
Everything works, thanks for the support

FloatingMind12, 2024-01-18:
I don't know if I should continue it here or post a new issue but I have encountered an issue where the cursor get stuck when I resize brush in krita.
I don't know if it's krita related but it worked with huion legacy driver
The huion legacy driver lags a lot so it's not an option

jamesbt365, 2024-01-18:
How are you resizing the brush?

FloatingMind12, 2024-01-18:
Shift + Drag

PR #3184 (2024-05-29): Add support for variant of Huion HS64.
Information collected from OpenTabletDriver/OpenTabletDriver#3183. This seems like a rebrand of the HS64 under a different name, but the name is obscure and hard to find any information about, any information about it just links back to the HS64, hence modifying the original configuration.

Fixes #3183.

All verification is on the linked issue.
closed 2024-01-18 by FloatingMind12