OpenTabletDriver/OpenTabletDriver #3187: Add support for tablet

xzczxczxccc, 2024-01-20:
### Manufacturer

Parblo

### Product Name

ninos n7b

### Device Width

248 мм

### Device Height

150 мм

### Pressure Levels

8192

### Lines per Inch (LPI)

5080 lpi

### Additional Information

There is a similar Parblo nison n7 tablet that is supported, but the n7b version is not supported, I can’t create the configuration myself :(

### Diagnostic Information

[121212.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13998548/121212.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-01-20:
Could you title your issue correctly instead of "Add support for tablet" thanks.

jamesbt365, 2024-06-14:
Fixed in #3006 bulk merged in #3314.

closed 2024-06-14 by jamesbt365