OpenTabletDriver/OpenTabletDriver #3189: Misdetecting "UGTABLET M708 V2" as "XP-Pen Star 03"

Kirtai, 2024-01-24:
### Steps to Reproduce

Plug a UGEE M708 V2 tablet in to my linux laptop.

### Expected Behavior

It should be detected as a UGEE M708 V2 and start working

### Observed Behavior

It was detected as an XP-Pen Star 03 and is unresponsive.

### Tablet Device

UGTABLET M708 V2

### Diagnostic Information

[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14041046/diag.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-03-11:
This is likely unresolvable unless the device strings are different, though the tablet should be working considering the links between UGEE and XP-Pen (UGEE owns XP-Pen), we already have documented the usage of one configuration for 2 distinct tablets because they are internally the same.

If it still isn't working, go to tablet > device string reader
10429
2311

dump all, save that file and send it here.

Kirtai, 2024-08-03:
Tried that, it didn't work.
Showed a dialogue window saying to replug the tablet but doing so and clicking ok just makes the dialogue appear again.

The tablet debugger in the git version (Arch AUR) shows it's detecting things but the main digitizer window just shows an error.

gonX, 2026-10-02:
There has been some changes for your tablet in recent times. v0.6.5.1 added a new identifier in #3646 and in v0.6.7's #4266 the configuration was further changed, so I'm going to assume this is fixed nowadays. Please reopen (or comment on) this issue if this issue still exists.

closed 2026-10-02 by gonX