OpenTabletDriver/OpenTabletDriver #3191: Support Request:

ccllyde, 2024-01-27:
### What do you need help with?

OpenTabletDriver is not detecting my Huion HS64 Tablet.

### What have you tried?

I've uninstalled all tablet drivers except OTD and I've restarted my computer.

### Diagnostic Information

[otd diagnostics.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14071254/otd.diagnostics.json)
{
  "App Version": "OpenTabletDriver v0.6.4.0",
  "Build Date": "12/22/2023",
  "Operating System": {
    "Name": "Win32NT",
    "Version": "10.0.22621.0",
    "Attributes": {
      "ServicePack": ""
    }
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\koe3n\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\koe3n\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\koe3n"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vhidkbms&col02#1&c6b730b&4&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "Unknown Product Name",
      "SerialNumber": "",
      "FriendlyName": "Unknown Product Name",
      "VendorID": 48879,
      "ProductID": 65260,
      "InputReportLength": 5,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false,
      "DeviceAttributes": {
        "HID_REPORTS": "03:0001:0002"
      }
    },
    {
      "DevicePath": "\\\\?\\hid#vhidkbms&col03#1&c6b730b&4&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "Unknown Product Name",
      "SerialNumber": "",
      "FriendlyName": "Unknown Product Name",
      "VendorID": 48879,
      "ProductID": 65260,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false,
      "DeviceAttributes": {
        "HID_REPORTS": "04:0001:0006"
      }
    },
    {
      "DevicePath": "\\\\?\\hid#vid_093a&pid_2530#6&1db1ba3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "PixArt",
      "ProductName": "Gaming Mouse",
      "SerialNumber": "",
      "FriendlyName": "Gaming Mouse",
      "VendorID": 2362,
      "ProductID": 9520,
      "InputReportLength": 9,
[... 1504 lines cut ...]
      "Time": "2024-01-27T00:55:45.2301996-05:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Wacom MTE-450'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-01-27T00:55:45.2302374-05:00",
      "Group": "Detect",
      "Message": "Searching for tablet 'Wacom PTH-450'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
   
I deleted the rest of the doc that I pasted here because I was getting an error "maximum is 65536 characters"

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-01-27:
device manager > view device by container > open your tablets container > screenshot

ccllyde, 2024-01-28:
[image]

jamesbt365, 2024-01-28:
right click Virtual HID Minidriver Collection > update > browse my computer for drivers > let me pick from a list of drivers on my computer > hid compliant device > next, replug your tablet.

closed 2024-01-28 by jamesbt365

ccllyde, 2024-01-28:
Holy shit thanks