OpenTabletDriver/OpenTabletDriver #3195: Cannot get diagnostic information on Fedora 39

azuredusk10, 2024-01-29:
### Steps to Reproduce

1. Go to Help > Export Diagnostics
2. See error message.

I've been able to reproduce this regardless of whether a tablet is plugged in or not, and regardless of whether it's a supported tablet or not.

### Expected Behavior

Be given the option to export the diagnostic information

### Observed Behavior

I get an error message that says
"An application error has occured. Report this to the developers!"

```
StreamJsonRpc.RemoteSerializationException: Error writing JSON RPC Message: FormatException: Input string was not in a correct format.
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync[TResult](RequestId id, String targetName, IReadOnlyList`1 arguments, IReadOnlyList`1 positionalArgumentDeclaredTypes, IReadOnlyDictionary`2 namedArgumentDeclaredTypes, CancellationToken cancellationToken, Boolean isParameterObject)
   at OpenTabletDriver.UX.MainForm.ExportDiagnostics() in /__w/OpenTabletDriver/OpenTabletDriver/dist/redhat/opentabletdriver/BUILD/opentabletdriver-0.6.4.0/OpenTabletDriver.UX/MainForm.cs:line 642
```

In the console, a new row is added, with the group name "RemoteSerializationException"

### Tablet Device

XP-PEN Deco Pro (Gen 2) XLW, and Wacom Intuos Pro M

### Diagnostic Information

No diagnostic information was generated

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

fybx, 2024-01-31:
Same error occurs both on Manjaro Linux and Arch Linux with kernels 6.5 and 6.7, X11 and Wayland.

jamesbt365, 2025-01-24:
Is this still an issue on the latest release of the driver, I have not seen this problem in long enough that I think its no longer an issue.

I couldn't reproduce it to begin with, so I can't test this myself.

gonX, 2025-01-25:
Based on the stack trace this looks like it's failing when asking Daemon to return the list of devices:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/9f4e357910a5917ef86166c195c858c1c65b46d2/OpenTabletDriver.UX/MainForm.cs#L642

Not sure how this fails, it returns `SerializedDeviceEndpoint` which has both an empty and an `IDeviceEndpoint` constructor. Stretch guess: maybe the `int`s in the interface need to be set as nullable?

gonX, 2025-08-17:
Thanks for the report @azuredusk10, but we are sadly still seeing reports of this bug on our Discord server.

closed 2025-12-18 by gonX