OpenTabletDriver/OpenTabletDriver #3196: Add support for Wacom Cintiq 13HDTouch model

Larfty, 2024-02-02:
### Manufacturer

Wacom

### Product Name

Wacom DTH-1300

### Device Width

298.45

### Device Height

165.1

### Pressure Levels

2048

### Additional Information

I was wondering if I could get support for my tablet. I see there is a non touch (DTK-1300) model that is supported however I have the touch model :( Any help is greatly appreciated!

### Diagnostic Information

[cintiqtest.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14145939/cintiqtest.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

Larfty, 2024-02-08:
Apologies if this has been asked already but is there a way for me to add my own config? I know it was possible in the older builds but I cannot figure it out in todays OTD.

jamesbt365, 2024-02-08:
You just create a Configurations folder next to the binaries.

Larfty, 2024-02-08:
[DTH-1300.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14214905/DTH-1300.json)

Alright cintiq 13hd touch version works now!

jamesbt365, 2024-02-08:
This should remain open until this is completely tested & a pull request is merged.

Larfty, 2024-02-08:
Oh sorry about that.