OpenTabletDriver/OpenTabletDriver #3204: Support Request:

xenomorph1096, 2024-02-09:
### What do you need help with?

Opentabletdriver detects my tablet (Wacom CTH-460), but the tablet does not work at all (fresh install)

### What have you tried?

1. Rebooting
2. `rmmod wacom`

### Diagnostic Information

{
  "AppVersion": "OpenTabletDriver v0.7.0.0-pre",
  "BuildDate": "",
  "OperatingSystem": {
    "Name": "Arch Linux",
    "Version": "Unknown",
    "Attributes": {
      "PRETTY_NAME": "Arch Linux",
      "ID": "arch",
      "BUILD_ID": "rolling",
      "KERNEL_VERSION": "6.7.2-arch1-1"
    }
  },
  "EnvironmentVariables": {
    "USER": "anant",
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-1",
    "PWD": "/home/anant",
    "PATH": "/usr/local/bin:/usr/bin"
  },
  "Devices": [
    {
      "VendorID": 1133,
      "ProductID": 2730,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "Manufacturer": "Logitech",
      "ProductName": "PRO X",
      "FriendlyName": "PRO X",
      "SerialNumber": "000000000000",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/0003:046D:0AAA.0003/hidraw/hidraw2",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "3",
        "HID_REPORT_DESCRIPTOR_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "VendorID": 1386,
      "ProductID": 209,
      "InputReportLength": 9,
      "OutputReportLength": 0,
      "FeatureReportLength": 17,
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "CTH-460",
      "FriendlyName": "CTH-460",
      "SerialNumber": "",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:00D1.000C/hidraw/hidraw0",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS": "01:0001:0002, 02:000D:0001"
      }
    },
    {
      "VendorID": 1386,
      "ProductID": 209,
      "InputReportLength": 20,
[... 1392 lines cut ...]
      "Time": "2024-02-09T11:08:57.8172088+01:00",
      "Group": "MouseBinding",
      "Message": "Button: Right",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-02-09T11:08:57.8174297+01:00",
      "Group": "Settings",
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

gonX, 2024-03-23:
Try resetting the configuration in the GUI:

File -> Reset to Defaults

If this does not work, I suggest that you use the release version of OpenTabletDriver (the `opentabletdriver` AUR package) and try with that instead.

Note that driver configuration between 0.6 and 0.7 is not interchangable and the driver currently fails silently if you have an incompatible configuration in place.

xenomorph1096, 2024-04-11:
Both of these do not seem to fix it for me. I reset to defaults, and also tried both the AUR packages, 0.6 and 0.7-git; resetting to defaults each time.

The tablet is detected, but it does not actually do anything; same as before. Very wierd. Is there anywhere else I should be looking for debug messages?
[image]

jamesbt365, 2024-04-11:
Your device has a flipped endpoint and won't work unless you override the configuration regardless of OTD version.

xenomorph1096, 2024-04-12:
A flipped endpoint? what is that? How do I override the configuration?

closed 2024-04-12 by jamesbt365

xenomorph1096, 2024-04-12:
Hey, thanks a lot! that totally worked for me. Maybe I missed this in the [wiki](https://opentabletdriver.net/Wiki/FAQ/Linux) somewhere, or maybe this is not mentioned. If it isn't it would be nice to add it to the wiki.

jamesbt365, 2024-04-12:
If we added a wiki entry for this tablet we would also need to add one for *every* configuration override we had, its just not worth the maintenance for the amount of tablets that may need configuration overrides, its easier to rework the detection scheme to accommodate these tablets #3011.