OpenTabletDriver/OpenTabletDriver #3205: Support Request: Gaomon 1060 pro not detected

HurtButton, 2024-02-10:
### What do you need help with?

The tablet can be used as a mouse but cannot be detected by OTD.
[image]

### What have you tried?

-Driver cleanup
-Restart computer
-Reinstall OTD
-Put "1060_Pro.json" into "Configuration"

### Diagnostic Information

[Diagnostic.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14228672/Diagnostic.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

HurtButton, 2024-02-12:
Solved by using a very old Gaomon driver (not the one on official website)
The problem may be due to versions
Virustotal: CrowdStrike Falcon: Win/grayware_confidence_60% (D)
[高漫1060.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14235033/1060.zip)

jamesbt365, 2024-03-11:
Tablet > device string reader
9580
110
dump all, send that file here.

HurtButton, 2024-03-17:
[String dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14625738/String.dump.txt)

SimonEast, 2024-03-21:
I've got a Gaomon 1060 Pro and it seems to be working *mostly* correctly with OTD v0.6.4.0, although I noticed it seems to be detected as a "Monoprice 10594" from looking at the status bar, which is not quite correct.

[image]

The GUI gives me 8 auxiliary bindings when the tablet actually has 12 hard buttons, and 16 numbers across the top which I suspect are some kind of software-enabled buttons?

It looks like this:

[image]

Is there a way to utilise those soft buttons?

Apart from the issues above, the tablet seems to work quite well despite the mid-detection.

Here's my string dump, which is almost identical to HurtButton's: [Gaomon 1060 Pro String dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14683641/Gaomon.1060.Pro.String.dump.txt)

SimonEast, 2024-03-21:
Actually I have noticed one other strange issue with the 1060 Pro, which I will add here:

I cannot get the pen buttons and pen tip (pressure) to trigger at the same time. When I open the tablet debugger at look at the raw output, I observe:

* Pressure levels change when pressing down on nib (✔ correct)
* Pressure remains at 0 when holding either pen button and pressing down on nib (⚠ seems strange, or is this normal?)
* `PenButtons:[False False]` with neither button pressed (✔ correct)
* `PenButtons:[True False]` with primary button pressed (✔ correct)
* `PenButtons:[False True]` with secondary button pressed (✔ correct)
* `PenButtons:[False True]` with *primary* button pressed and pressing down on nib (❌ strange and incorrect)
* `PenButtons:[False False]` with secondary button pressed and pressing down on nib (❌ strange and incorrect)

Is this an issue with OTD configuration, or perhaps a deeper issue with the driver itself? When I look in Windows device manager I have 1060PRO driver version 8.33.30.0 from April 2014. Not sure if there's a newer one than that.

If this should be moved to a separate Github issue, let me know.

jamesbt365, 2024-03-21:
You can tell if its normal by looking at the bytes that are responsible for pressure while pressing the pen buttons.

gonX, 2026-10-02:
v0.6.7 had a new identifier added for this tablet in #4666 - can you check if this fixed in v0.6.7?