OpenTabletDriver/OpenTabletDriver #3214: ScriptRunner crash driver when binded to a button

noigamegun, 2024-02-17:
### Steps to Reproduce

1. bind a button to run a bat file
2. apply and save everything
3. driver crashed when button pressed

### Expected Behavior

Run the bat file

### Observed Behavior

Driver crashed with a error of
``` [MissingMethodException:Error]	Method not found: 'Void OpenTabletDriver.Plugin.Log.Exception(System.Exception)'.
   at VoiDPlugins.Binding.ScriptRunner.ScriptRunner.Press(TabletReference tablet, IDeviceReport report)
   at OpenTabletDriver.Desktop.Binding.BindingState.Invoke(TabletReference tablet, IDeviceReport report, Boolean newState) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Desktop\Binding\BindingState.cs:line 19
   at OpenTabletDriver.Desktop.Binding.BindingHandler.HandleBindingCollection(TabletReference tablet, IDeviceReport report, IDictionary`2 bindings, IList`1 newStates) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Desktop\Binding\BindingHandler.cs:line 80
   at OpenTabletDriver.Desktop.Binding.BindingHandler.HandleAuxiliaryReport(TabletReference tablet, IAuxReport report) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Desktop\Binding\BindingHandler.cs:line 64
   at OpenTabletDriver.Desktop.Binding.BindingHandler.HandleBinding(IDeviceReport report) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Desktop\Binding\BindingHandler.cs:line 46
   at OpenTabletDriver.Desktop.Binding.BindingHandler.Consume(IDeviceReport report) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Desktop\Binding\BindingHandler.cs:line 37
   at Pressure_Debounce.Pressure_Debounce.Consume(IDeviceReport value)
   at OpenTabletDriver.Plugin.Output.OutputMode.Consume(IDeviceReport report) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Plugin\Output\OutputMode.cs:line 113
   at OpenTabletDriver.Plugin.Output.AsyncPositionedPipelineElement`1.OnEmit() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Plugin\Output\AsyncPositionedPipelineElement.cs:line 121
   at TabletDriverFilters.Devocub.Antichatter.ConsumeState()
   at OpenTabletDriver.Plugin.Output.AsyncPositionedPipelineElement`1.Consume(T value) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Plugin\Output\AsyncPositionedPipelineElement.cs:line 76
   at OpenTabletDriver.Plugin.Output.OutputMode.Read(IDeviceReport deviceReport) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Plugin\Output\OutputMode.cs:line 115
   at OpenTabletDriver.InputDeviceTree.HandleReport(Object sender, IDeviceReport report) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\InputDeviceTree.cs:line 63
   at OpenTabletDriver.Devices.DeviceReader`1.OnReport(T report) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Devices\DeviceReader.cs:line 140
   at OpenTabletDriver.Devices.DeviceReader`1.Main() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver\Devices\DeviceReader.cs:line 111```

### Tablet Device

Veikk VK430

### Diagnostic Information

[ScriptRunner.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14317783/ScriptRunner.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

noigamegun, 2024-02-17:
Downloading the plugin off the website reproduced the same bug

Here is what the bug look like

https://github.com/OpenTabletDriver/OpenTabletDriver/assets/76461912/ff6177f3-21f1-4a1b-a743-64dc43d8aa92

jamesbt365, 2024-02-17:
Issues with plugins should be reported to the repository of the specific plugin, plugins are extensions to the driver and are not contained within this repository. You can find the repository for this plugin when you press "Show source code" in the plugin manager.

closed 2024-02-17 by jamesbt365