OpenTabletDriver/OpenTabletDriver #3216: XP-Pen Artist 10 (2nd Gen) : Multi-Monitor : Incorrect Drawing Area

guspower, 2024-02-19:
### Steps to Reproduce

### Set Up:
* Linux `6.6.13-gentoo`
* i3
* XP-Pen Artist 10 (2nd Gen) 
* `xrandr --output DisplayPort-2 --right-of HDMI-A-0`

### Display Configuration
```
Screen 0: minimum 320 x 200, current 4480 x 1080, maximum 16384 x 16384
HDMI-A-0 connected primary 2560x1080+0+0 (normal left inverted right x axis y axis) 798mm x 334mm
   2560x1080     60.00*+
   1920x1080     60.00    50.00    59.94    30.00    29.97  
   1680x1050     59.88  
   1600x900      60.00  
   1280x1024     75.02    60.02  
   1440x900      60.00  
   1280x800      60.00  
   1152x864      75.00  
   1280x720      60.00    50.00    59.94  
   1024x768      75.03    60.00  
   800x600       75.00    60.32  
   720x576       50.00  
   720x480       60.00    59.94  
   640x480       75.00    60.00    59.94  
   720x400       70.08  
DisplayPort-0 disconnected (normal left inverted right x axis y axis)
DisplayPort-1 disconnected (normal left inverted right x axis y axis)
DisplayPort-2 connected 1920x1080+2560+0 (normal left inverted right x axis y axis) 476mm x 268mm
   1920x1080     59.90*+
   1680x1050     59.88  
   1280x1024     60.02  
   1440x900      59.90  
   1360x768      59.95  
   1280x800      59.90  
   1280x720      60.00  
   1024x768      60.00  
   1152x648      60.01  
   800x600       60.32  
   640x480       59.94 
```

### Run Daemon
```
$ ./OpenTabletDriver.Daemon 
[Detect:Info]	Searching for tablets...
[Detect:Debug]	Searching for tablet '10moon 1060N'
...
[Detect:Debug]	Searching for tablet 'XP-Pen Artist 10 (2nd Gen)'
[Device:Debug]	Initializing device 'Artist 10 (2nd Gen)' /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.4/usb5/5-1/5-1.5/5-1.5:1.2/0003:28BD:094D.000B/hidraw/hidraw5
[Device:Debug]	Using report parser type 'OpenTabletDriver.Configurations.Parsers.XP_Pen.XP_PenOffsetPressureReportParser'
[Device:Debug]	Set device output: 02-B0-04
[Detect:Info]	Found tablet 'XP-Pen Artist 10 (2nd Gen)'
...
Evdev:Debug]	Successfully initialized virtual tablet. (code NONE)
[XP-Pen Artist 10 (2nd Gen):Info]	Output mode: Absolute Mode
[XP-Pen Artist 10 (2nd Gen):Info]	Display area: [2560x1080@<1280, 540>:0°],
[XP-Pen Artist 10 (2nd Gen):Info]	Tablet area: [224.51x126.695@<225, 126>:0°],
[XP-Pen Artist 10 (2nd Gen):Info]	Clipping: Enabled
[XP-Pen Artist 10 (2nd Gen):Info]	Ignoring reports outside area: Disabled
[XP-Pen Artist 10 (2nd Gen):Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[XP-Pen Artist 10 (2nd Gen):Info]	Filters: OpenTabletDriver.Desktop.Binding.BindingHandler
[Settings:Info]	Driver is enabled.
```

### Expected Behavior

### Expected

Pen should map to tablet screen.

### Observed Behavior

### Actual

Pen maps to half tablet screen; 
* the cursor and pen line up on the right hand edge
* as the pen moves left the cursor runs ahead of it
* the cursor is at the left hand edge when the pen is in the middle of the screen

### Tablet Device

XP-Pen Artist 10 (2nd Gen)

### Diagnostic Information

[xppen-10-diagnostics.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14332235/xppen-10-diagnostics.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

LuisIllig, 2024-03-05:
I have the same issues with XP-Pen Artist 13 (2nd Gen).

jamesbt365, 2024-03-05:
Both of y'all display mappings are wrong or you have your window manager mapping the tablet too, you should not be double mapping the tablet.

Right click the display mapping in OTD and set to display.

guspower, 2024-03-11:
Hi @jamesbt365 - thanks for the response. 

Could you please elaborate further on the solution? 

At the moment I perform the following steps:

1. Connect the tablet using the 3-in-1 usb connector
    - The X server detects the tablet as a `HANVON UGEE Artist 10 (2nd Gen)` (xorg log output below)
    - The screen on the tablet switches on and shows a virtual desktop
    - `xrandr` shows that it is connected on `DisplayPort-2` (connected 1920x1080+2560+0 (normal left inverted right x axis y axis) 530mm x 290mm)
2. Run the `otd-daemon`: `[Detect:Info]	Found tablet 'XP-Pen Artist 10 (2nd Gen)'`
3. Run `./OpenTabletDriver.UX.Gtk`: it shows the display at the top and tablet at the bottom. 

When you say:
1. display mappings are wrong or 
2. you have your window manager mapping the tablet too, you should not be double mapping the tablet

Can you say more about specifically what I need to change in my setup?

### xorg log
```
[383605.946] (II) config/udev: Adding input device HANVON UGEE Artist 10 (2nd Gen) (/dev/input/event11)
[383605.946] (**) HANVON UGEE Artist 10 (2nd Gen): Applying InputClass "evdev touchpad catchall"
[383605.946] (**) HANVON UGEE Artist 10 (2nd Gen): Applying InputClass "libinput touchpad catchall"
[383605.946] (II) Using input driver 'libinput' for 'HANVON UGEE Artist 10 (2nd Gen)'
[383605.998] (II) systemd-logind: got fd for /dev/input/event11 13:75 fd 56 paused 0
[383605.998] (**) HANVON UGEE Artist 10 (2nd Gen): always reports core events
[383605.998] (**) Option "Device" "/dev/input/event11"
[383605.998] (II) event11 - failed to create input device '/dev/input/event11'.
[383605.998] (EE) libinput: HANVON UGEE Artist 10 (2nd Gen): Failed to create a device for /dev/input/event11
[383605.998] (EE) PreInit returned 2 for "HANVON UGEE Artist 10 (2nd Gen)"
[383605.998] (II) UnloadModule: "libinput"
[383605.998] (II) systemd-logind: releasing fd for 13:75
[383621.811] (II) config/udev: Adding input device OpenTabletDriver Virtual Tablet (/dev/input/event12)
[383621.811] (**) OpenTabletDriver Virtual Tablet: Applying InputClass "evdev pointer catchall"
[383621.811] (**) OpenTabletDriver Virtual Tablet: Applying InputClass "libinput pointer catchall"
[383621.811] (II) Using input driver 'libinput' for 'OpenTabletDriver Virtual Tablet'
[383621.812] (II) systemd-logind: got fd for /dev/input/event12 13:76 fd 58 paused 0
[383621.812] (**) OpenTabletDriver Virtual Tablet: always reports core events
[383621.812] (**) Option "Device" "/dev/input/event12"
[383621.813] (II) event12 - OpenTabletDriver Virtual Tablet: is tagged by udev as: Mouse
[383621.814] (II) event12 - OpenTabletDriver Virtual Tablet: device is a pointer
[383621.814] (II) event12 - OpenTabletDriver Virtual Tablet: device removed
[383621.814] (**) Option "config_info" "udev:/sys/devices/virtual/input/input33/event12"
[383621.814] (II) XINPUT: Adding extended input device "OpenTabletDriver Virtual Tablet" (type: MOUSE, id 16)
[383621.814] (**) Option "AccelerationScheme" "none"
[383621.814] (**) OpenTabletDriver Virtual Tablet: (accel) selected scheme none/0
[383621.814] (**) OpenTabletDriver Virtual Tablet: (accel) acceleration factor: 2.000
[383621.814] (**) OpenTabletDriver Virtual Tablet: (accel) acceleration threshold: 4
[383621.815] (II) event12 - OpenTabletDriver Virtual Tablet: is tagged by udev as: Mouse
[383621.816] (II) event12 - OpenTabletDriver Virtual Tablet: device is a pointer
```

Thanks for your time - appreciate it!

jamesbt365, 2024-03-11:
I want a screenshot of the OTD interface and the diagnostic file from help > export diagnostic, the xorg log unfortunately isn't all that helpful.

rmartinsanchez, 2024-06-12:
> Both of y'all display mappings are wrong or you have your window manager mapping the tablet too, you should not be double mapping the tablet.
> 
> Right click the display mapping in OTD and set to display.

I have tried this but it doesn't solve the problem. I currently use XP-Pen Artist 12 Pro.

I use 3 monitors. All with the same resolution.

This is my diagnostic file.

[diagnostic.json](https://github.com/user-attachments/files/15810699/diagnostic.json)

Thanks for the great work!

jamesbt365, 2024-06-12:
@rmartinsanchez What Wayland compositor?

rmartinsanchez, 2024-06-12:
I use manjaro with KDE. From what I understand it uses kwin.

The driver that comes by default has turned my axes upside down and the configuration options are not enough to correct the problem. 

This driver (opentabletdriver) does almost everything I need except for that small issue with mapping. Since when the pen is in the upper left corner, the cursor is also there. But when I get to the center of the tablet area with my pen, the cursor is at the bottom right corner of my tablet.

I hope this information helps.

jamesbt365, 2024-06-12:
This is an upstream issue with KDE Wayland, it does not let you disable its inbuilt tablet mapping.

guspower, 2024-06-13:
@rmartinsanchez  thanks for bumping this!

@jamesbt365 screenshot and diagnostic attached.

[xppen-10-diagnostics.json](https://github.com/user-attachments/files/15818509/xppen-10-diagnostics.json)

[image]

jamesbt365, 2024-06-13:
Tablet > tablet debugger, move your pen to the bottom right of the tablet and get the highest x and y coordinates you can.

guspower, 2024-06-13:
`44902, 25339` ?

[image]

jamesbt365, 2024-06-13:
Could you send your I3 configuration?

guspower, 2024-06-13:
```
set $mod Mod4
font pango:monospace 8
exec --no-startup-id dex --autostart --environment i3
exec --no-startup-id xss-lock --transfer-sleep-lock -- i3lock --nofork
exec --no-startup-id nm-applet
set $refresh_i3status killall -SIGUSR1 i3status
bindsym XF86AudioRaiseVolume exec --no-startup-id pactl set-sink-volume @DEFAULT_SINK@ +10% && $refresh_i3status
bindsym XF86AudioLowerVolume exec --no-startup-id pactl set-sink-volume @DEFAULT_SINK@ -10% && $refresh_i3status
bindsym XF86AudioMute exec --no-startup-id pactl set-sink-mute @DEFAULT_SINK@ toggle && $refresh_i3status
bindsym XF86AudioMicMute exec --no-startup-id pactl set-source-mute @DEFAULT_SOURCE@ toggle && $refresh_i3status
floating_modifier $mod
tiling_drag modifier titlebar
bindsym $mod+Return exec i3-sensible-terminal
bindsym $mod+Shift+q kill
bindsym $mod+d exec "rofi -show run"
bindsym $mod+g exec "rofi -show window"
bindsym Mod1+Tab exec "rofi -show window"
bindsym $mod+j focus left
bindsym $mod+k focus down
bindsym $mod+l focus up
bindsym $mod+semicolon focus right
bindsym $mod+Left focus left
bindsym $mod+Down focus down
bindsym $mod+Up focus up
bindsym $mod+Right focus right
bindsym $mod+Shift+j move left
bindsym $mod+Shift+k move down
bindsym $mod+Shift+l move up
bindsym $mod+Shift+semicolon move right
bindsym $mod+Shift+Left move left
bindsym $mod+Shift+Down move down
bindsym $mod+Shift+Up move up
bindsym $mod+Shift+Right move right
bindsym $mod+h split h
bindsym $mod+v split v
bindsym $mod+f fullscreen toggle
bindsym $mod+s layout stacking
bindsym $mod+w layout tabbed
bindsym $mod+e layout toggle split
bindsym $mod+Shift+space floating toggle
bindsym $mod+space focus mode_toggle
bindsym $mod+a focus parent
set $ws1 "1"
set $ws2 "2"
set $ws3 "3"
set $ws4 "4"
set $ws5 "5"
set $ws6 "6"
set $ws7 "7"
set $ws8 "8"
set $ws9 "9"
set $ws10 "10"
bindsym $mod+1 workspace number $ws1
bindsym $mod+2 workspace number $ws2
bindsym $mod+3 workspace number $ws3
bindsym $mod+4 workspace number $ws4
bindsym $mod+5 workspace number $ws5
bindsym $mod+6 workspace number $ws6
bindsym $mod+7 workspace number $ws7
bindsym $mod+8 workspace number $ws8
bindsym $mod+9 workspace number $ws9
bindsym $mod+0 workspace number $ws10
bindsym $mod+Shift+1 move container to workspace number $ws1
bindsym $mod+Shift+2 move container to workspace number $ws2
bindsym $mod+Shift+3 move container to workspace number $ws3
bindsym $mod+Shift+4 move container to workspace number $ws4
bindsym $mod+Shift+5 move container to workspace number $ws5
bindsym $mod+Shift+6 move container to workspace number $ws6
bindsym $mod+Shift+7 move container to workspace number $ws7
bindsym $mod+Shift+8 move container to workspace number $ws8
bindsym $mod+Shift+9 move container to workspace number $ws9
bindsym $mod+Shift+0 move container to workspace number $ws10
bindsym $mod+Shift+c reload
bindsym $mod+Shift+r restart
bindsym $mod+Shift+e exec "i3-nagbar -t warning -m 'You pressed the exit shortcut. Do you really want to exit i3? This will end your X session.' -B 'Yes, exit i3' 'i3-msg exit'"
bindsym $mod+Shift+n move scratchpad
bindsym $mod+n scratchpad show
mode "resize" {
        bindsym j resize shrink width 10 px or 10 ppt
        bindsym k resize grow height 10 px or 10 ppt
        bindsym l resize shrink height 10 px or 10 ppt
        bindsym semicolon resize grow width 10 px or 10 ppt
        bindsym Left resize shrink width 10 px or 10 ppt
        bindsym Down resize grow height 10 px or 10 ppt
        bindsym Up resize shrink height 10 px or 10 ppt
        bindsym Right resize grow width 10 px or 10 ppt
        bindsym Return mode "default"
        bindsym Escape mode "default"
        bindsym $mod+r mode "default"
}
bindsym $mod+r mode "resize"
bar {
        status_command i3status
}
```

gonX, 2026-10-02:
I'm going to close this support request as stale as this sounds like either #1143 or your desktop environment (or X11) remapping the area for us, the latter of which has recently been covered in [the FAQ on our website](https://opentabletdriver.net/Wiki/FAQ/Linux#multi-monitor-mapping-issues).

closed 2026-10-02 by gonX