OpenTabletDriver/OpenTabletDriver #3218: Screen sizing is incorrect on multi-monitor setups

Sollace, 2024-02-22:
### Steps to Reproduce

I've found there is some problem with how the screen position is being calculated when going from tablet coordinates to screen coordinates.

For reproduction, my setup is:

1. Top-left HP external monitor at 2560x1440 @ 125%
2. Top-right HP external display port monitor at 2560x1440 @ 125%
3. Bottom-left Huion KAMVAS 13 (model GS1331) at 1920x1080 @ 100%

[image]
[image]

# System Specifications

Operating System: Manjaro Linux 
KDE Plasma Version: 5.27.10
KDE Frameworks Version: 5.115.0
Qt Version: 5.15.12
Kernel Version: 6.6.16-2-MANJARO (64-bit)
Graphics Platform: Wayland
Processors: 16 × 11th Gen Intel® Core™ i9-11950H @ 2.60GHz
Memory: 31.1 GiB of RAM
Graphics Processor: Mesa Intel® UHD Graphics
Manufacturer: HP
Product Name: HP ZBook Fury 17.3 inch G8 Mobile Workstation PC

### Expected Behavior

Setting the area to tablet's display area should match (at least roughly) to the tablet display's dimensions.

### Observed Behavior

If I set the display area to just the tablet area, the cursor appears only in one quarter region of the tablet. If I set it to the full virtual display area (so spanning all displays) the cursor covers the full tablet display, but the coordinates don't match exactly and it starts to lag behind the pen on the X or Y axis when moving to the bottom and right.

Through trial and error, and manually tweaking the width and height, I eventually landed onto a region that properly matches to my tablet's display, and it looks like this:

[image]

### Tablet Device

Huion KAMVAS 13 (model GS1331)

### Diagnostic Information

PLACEHOLDFER

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-02-22:
This is a KDE Wayland issue, it does not allow using the tablet on the entire virtual desktop, so both OTD and KDE are trying to map the tablet. Full artist mode passthrough is planned, and is tracked in another issue I think. This issue is also a duplicate of another.

You have a few potentional "solutions":
1 - Use X11
2 - Use a different wayland compositor.
3 - Don't use OTD and instead use the KDE controls.

closed 2024-02-22 by jamesbt365

Sollace, 2024-02-22:
> This is a KDE Wayland issue, it does not allow using the tablet on the entire virtual desktop, so both OTD and KDE are trying to map the tablet. Full artist mode passthrough is planned, and is [...]

Alright thanks.

KDE controls didn't work for me before, so I will continue with what I have. Thanks though.

rkolbaskin, 2024-04-21:
KWin is trying to manage mapping while OTD already does it. KDE devs already added an option to not do this into KWin, but not into configuration app.

You can either tell OTD to map your tablet to full workspace (Set to display -> Virtual Display) then select the tablet screen in kcm, or run `qdbus org.kde.KWin /org/kde/KWin/InputDevice/event<n> org.kde.KWin.InputDevice.mapToWorkspace 1` to switch the option mentioned above. Replace `event<n>` with "OpenTabletDriver Virtual Artist Tablet" event id you can find with `evtest`

lboklin, 2024-07-03:
I have this problem on X11 with an ultrawide (3440x1440) monitor and xp-pen artist 15.6 pro, so it's not a wayland issue (or it's a separate underlying cause from OP).

lboklin, 2024-07-05:
I also want to add that there is a *rotation* problem too, likely because of calculations involving fractions (to get rect centre?) of incorrect dimensions

rupert-jones-99, 2025-09-06:
Happens on Hyprland too btw