OpenTabletDriver/OpenTabletDriver #3222: Add support for mapping all Cintiq 12WX Auxiliary buttons

Megalomaniak, 2024-02-29:
### Manufacturer

Wacom

### Product Name

DTZ-1200W

### Additional Information

The Wacom Cintiq 12 WX currently falls back to using Intuos 3 Configuration best that I can tell. Which mostly works, except unlike the Intuos 3 which has only 8 Auxiliary buttons total (4 on each side) the Cintiq 12 WX features 10 total, 5 on each side.

Opening the Debugger and pressing the 2 missing buttons I see a change in RAW values so I recon it should be possible to support them?

Whether to resolve this by adding the configurability of the 2 additional buttons via the Intuos 3 config or by giving the Cintiq 12 WX it's own unique config I'm not sure. Intuition says latter would be more appropriate however since the devices are so similar perhaps the former makes more sense in terms of technical debt.

### RAW values for the 2 missing buttons

**Report Type:** OpenTabletDriver.Configurations.Parsers.Wacom.Intuos3.Intuos3Aux-Report

```
0C 00 00 00 00 10 00 00 00 00 // topmost button on left
0C 00 00 00 00 00 10 00 00 00 // topmost button on right
```

and as an example of Aux button that is mappable, rest of the buttons on right:
```
0C 00 00 00 00 00 01 00 00 00
0C 00 00 00 00 00 02 00 00 00
0C 00 00 00 00 00 04 00 00 00
0C 00 00 00 00 00 08 00 00 00
```

---

While at it perhaps it's also worth a mention that similar to Intuos 3 the 12WX has touch strips on each side and I can see value changes in debugger also from their input.

### Touchstrip RAW value ranges

**Report Type:** OpenTabletDriver.Configurations.Parsers.Wacom.Intuos3.Intuos3Aux-Report
Values are obtained by sliding the finger over the strip from top to bottom, 1 state per line

##### Left touch strip values 

```
0C 00 01 00 00 00 00 00 00 00
0C 00 02 00 00 00 00 00 00 00
0C 00 04 00 00 00 00 00 00 00
0C 00 08 00 00 00 00 00 00 00
0C 00 10 00 00 00 00 00 00 00
0C 00 20 00 00 00 00 00 00 00
0C 00 40 00 00 00 00 00 00 00
0C 00 80 00 00 00 00 00 00 00
0C 01 00 00 00 00 00 00 00 00
0C 02 00 00 00 00 00 00 00 00
0C 04 00 00 00 00 00 00 00 00
0C 08 00 00 00 00 00 00 00 00
0C 10 00 00 00 00 00 00 00 00
```

And same but 2 column positions over to the right for right strip

##### Right touch strip values

```
0C 00 00 00 01 00 00 00 00 00
            ...
0C 00 00 10 00 00 00 00 00 00
```

I'm mostly concerned with the buttons however I can see the value in being able to use the touch strips in something like krita for brush size or canvas rotation for an example. But mapping these functions to the strips should probably be done in the respective application settings rather than the driver config since it's application specific. So that might have to be a feature request for a tracker elsewhere I'm guessing. Woe unto me, Krita is part of KDE family and they use a common bugzilla instance for all of them. Ugh.

**Let me know if I can provide any extra feedback.**

### Diagnostic Information

File is too large for github so I've created a public gist:

https://gist.github.com/Megalomaniak/b93383301b8dc589ca4ce59da0fef054

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-03-12:
@Megalomaniak could you test the action build [here](https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/8247185599) please? Downloads are at the bottom. Source code is available in #3241 and #3242 if you would like to see or compile the changes yourself.

Megalomaniak, 2024-03-13:
@jamesbt365 Since I'm on linux I presume I need both the linux build and the udev rules? Or is the build alone enough?

jamesbt365, 2024-03-13:
Aaa I love the website on mobile jumping around!

If you are keeping a packaged version of OTD installed you dont need to download the rules, if you are compiling a packaged build they'll come included,if you dont have OTD installed yes, youll need the rules.

Megalomaniak, 2024-03-13:
OK, I'm normally using OTD-git from Aur so I guess I just need the binary then, now if only I could find some install instructions for this. Or maybe I really should try git pull and build...

jamesbt365, 2024-03-13:
I'm highly against the usage of OTD-git, there is absolutely no benefits of using it over a release build and it only brings more issues to the table, it runs on a different codebase which is way more unstable and lacking features as of right now, alongside a bunch of regressions & other bugs.

However, if you still want to do this nothing is stopping you from doing these modifications to the `opentabletdriver-git` package build.

Here is a patched PKGBUILD that pulls from a local tar instead of pulling it from a github release, you can compile that bit yourself if you wish too.

Here is a file you can directly install with `pacman -U` (after you remove the .txt at the end of the filename).
[opentabletdriver-0.6.4.0-2-x86_64.pkg.tar.zst.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14592299/opentabletdriver-0.6.4.0-2-x86_64.pkg.tar.zst.txt)

Or you can run the packaging scripts `./eng/linux/package.sh --package BinaryTarBall` to build a binary for 0.6.x yourself off #3242, do not do this on master as it will require more changes, you are better off modifying the pkgbuild for opentabletdriver-git to point at the branch used in #3241 at that point.

Then move the created tar.gz next to a pkgbuild with the source set to `file://relative_path_to_tar.gz`, then `makepkg -si`

If you are insistent on using a build based off `master` you can just modify the source of the opentabletdriver-git package to point at that above as i previously mentioned, that should be the only change.

jamesbt365, 2024-03-13:
You likely never could get it to work because you already had 0.7 settings, which are not compatible with 0.6.x, remove the ~/.config/OpenTabletDriver directory.

Megalomaniak, 2024-03-13:
No I meant I had tried 0.6.x before I found the aur for 0.7.

Megalomaniak, 2024-03-13:
`6.4.0+7b98ae8115e65adb19804deaf47ce73bc99b5233` Still only provides me 8 Aux buttons to configure and in Krita trying to draw with the pen has a random chance results in grabbing and panning the canvas instead of drawing. Maybe once in every 19 strokes on average. When it does draw at least pressure does work.

jamesbt365, 2024-03-13:
May bad, uploaded the wrong file and instead gave you a release build.
[opentabletdriver-0.6.4.0-x64.tar.gz.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14593090/opentabletdriver-0.6.4.0-x64.tar.gz.txt)

jamesbt365, 2024-03-13:
> Krita trying to draw with the pen has a random chance it results in grabbing and panning the canvas instead of drawing. Maybe once in every 19 strokes on average. When it does draw at least pressure does work.

The parsing and pipeline is functionally the same as 0.7, so nothing different should be happening here, could you check the tablet > tablet debugger to see if its happening there?

if it isn't, its probably a bug with krita because OTDs virtual tablet is nothing different to a real tablet without any passthrough.

Megalomaniak, 2024-03-13:
Are we certain it is the correct one this time. As in should I rename this latest one to `*.pkg.tar.zst` or unpack and then makepkg myself?

jamesbt365, 2024-03-13:
Its the right one this time.

Megalomaniak, 2024-03-14:
I asked because the tarball is missing package metadata so `sudo pacman -U opentabletdriver-0.6.4.0-x64.tar.gz` predictably gives me an `error: missing package metadata in opentabletdriver-0.6.4.0-x64.tar.gz`

jamesbt365, 2024-03-14:
I... uh, renamed and uploaded the wrong file...

Why do they have to be called such similar things, shame on me for not double checking.

[opentabletdriver-0.6.4.0-2-x86_64.pkg.tar.zst.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14595387/opentabletdriver-0.6.4.0-2-x86_64.pkg.tar.zst.txt)

Megalomaniak, 2024-03-14:
So far seems to be the same as the first one you uploaded. I made sure to download the newlinked one as `*.new` instead of `*.txt` then renamed it to remove that ending and after installing it the opentabletdriver help > about still reports the same version string: `v0.6.4.0+7b98ae8115e65adb19804deaf47ce73bc99b5233` as last time.

And only 8 Aux buttons show up to be clear.

jamesbt365, 2024-03-14:
Well then I either fucked up for 1000th time at 1am with my 1000 folders named OpenTabletDriver which is very likely or its something else.

You are welcome to attempt this yourself with the PKGBUILD from the aur and building off the backport PR, or just running off the tarball in the action builds. I'll look at this thread when I wake up.

Megalomaniak, 2024-03-14:
> Well then I either fucked up for 1000th time at 1am with my 1000 folders named OpenTabletDriver which is very likely or its something else.

Yeah.... I've been suspecting for a while now that since you tried to upload a file with the same exact name it(that being github) decided to drop the file upload and instead attached a link to the exact same file in that latest post. If that makes sense...

Megalomaniak, 2024-03-14:
OK, I've copied the PKGBUILD from the aur:

```
# Maintainer: Sebastian 'gonX' Jensen <gonx@gonx.dk>
# Contributor: LavaDesu <me@lava.moe>
pkgname=opentabletdriver
_pkgname=OpenTabletDriver
_lpkgname=opentabletdriver
_spkgname=otd
pkgver=0.6.4.0
pkgrel=3
pkgdesc="A cross-platform open source tablet driver"
arch=('x86_64')
url="https://opentabletdriver.net"
license=('LGPL3')
depends=('dotnet-runtime-6.0' 'gtk3' 'libevdev')
optdepends=('libxrandr: x11 display querying support' 'libx11')
makedepends=('dotnet-sdk>=6.0' 'jq' 'git')
conflicts=('digimend-kernel-drivers-dkms-git' 'digimend-drivers-git-dkms' 'digimend-kernel-drivers-dkms' 'digimend-kernel-drivers')
install="notes.install"
# unified binary dotnet releases break when stripped see https://github.com/dotnet/runtime/issues/54947
options=('!strip')
source=("OpenTabletDriver-$pkgver.tar.gz::https://github.com/OpenTabletDriver/OpenTabletDriver/archive/v$pkgver.tar.gz"
        "$_lpkgname.desktop"
        "notes.install"
        )

sha256sums=('SKIP')

_srcdir="OpenTabletDriver-$pkgver"

# TODO: use eng/lib.sh
#prepare() {
#    cd "$srcdir"
#    source eng/lib.sh
#}

build() {
    export DOTNET_CLI_TELEMETRY_OPTOUT=1
    export DOTNET_SKIP_FIRST_TIME_EXPERIENCE=true

    cd "$srcdir/$_srcdir"

    if check_option "strip" y; then
        EXTRA_OPTIONS="/p:DebugType=None /p:DebugSymbols=false"
    fi

    ./eng/linux/package.sh -- $EXTRA_OPTIONS

    OTD_CONFIGURATIONS="${PWD}/OpenTabletDriver.Configurations/Configurations" ./generate-rules.sh > 70-$_lpkgname.rules
}

package() {
    cd "$srcdir"

    sed -i "s/OTD_VERSION/$pkgver/" "$_lpkgname.desktop"

    install -Dm 644 -o root "$_srcdir/70-$_lpkgname.rules" -t "$pkgdir/usr/lib/udev/rules.d"
    install -Dm 644 -o root "$_srcdir/$_pkgname.UX/Assets/$_spkgname.png" -t "$pkgdir/usr/share/pixmaps"

    install -Dm 755 -o root -t "$pkgdir/usr/bin" \
                    "$_srcdir/eng/linux/Generic/usr/bin/$_spkgname" \
                    "$_srcdir/eng/linux/Generic/usr/bin/$_spkgname-daemon" \
                    "$_srcdir/eng/linux/Generic/usr/bin/$_spkgname-gui"

    install -Dm 755 -o root -t "$pkgdir/usr/lib/$_lpkgname" \
                    "$_srcdir/dist/$_pkgname.Console" \
                    "$_srcdir/dist/$_pkgname.Daemon" \
                    "$_srcdir/dist/$_pkgname.UX.Gtk"

    install -Dm 644 -o root "$_srcdir/eng/linux/Generic/usr/lib/systemd/user/$_lpkgname.service" -t "$pkgdir/usr/lib/systemd/user"
    install -Dm 644 -o root "$_srcdir/eng/linux/Generic/usr/lib/modprobe.d/99-$_lpkgname.conf" -t "$pkgdir/usr/lib/modprobe.d"
    install -Dm 644 -o root "$_srcdir/eng/linux/Generic/usr/lib/modules-load.d/$_lpkgname.conf" -t "$pkgdir/usr/lib/modules-load.d"
    install -Dm 644 -o root "$_lpkgname.desktop" -t "$pkgdir/usr/share/applications"
    install -Dm 644 -o root "$_srcdir/docs/manpages/$_lpkgname.8" -t "$pkgdir/usr/share/man/man8"
}

```

Replaced the SHA256sums with SKIP, but other than that, how to actually get the exact branch and pull request needed from github is still a mystery to me.

PR #3241 (2024-05-29): Add support for Wacom Cintiq 12WX (DTZ-1200W) 5th and 10th buttons
Fixes #3222.

Data recordings and anything relevant in within that issue.

A new parser is required to keep ordering of the old buttons, maybe a better way exists so that I don't have to duplicate the whole parser.

The way Intuos3 Aux works *typically* is 4 buttons per side and the byte that the data is different per side, this is why a new parser is needed to not break existing aux.

Alternatively a parser change could be made without creating a new parser but this would mean that the 5th and 10th buttons are the last to be b [... cut]
closed 2024-05-29 by InfinityGhost