OpenTabletDriver/OpenTabletDriver #3223: Add support for

ORI-Alnitak, 2024-03-01:
### Manufacturer

gaomon

### Product Name

WH851

### Device Width

288mm

### Device Height

170mm

### Pressure Levels

16384

### Lines per Inch (LPI)

5080

### Diagnostic Information

```json
[1.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14460499/1.json)
```

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-03-01:
Change the name of the title so it actually contains the name of the tablet please.

And uninstall the gaomon driver then export another diagnostic, we cannot proceed otherwise.

jamesbt365, 2024-03-03:
You can modify the title of an issue without creating a new issue, but whatever. Closing in favour of #3224.

closed 2025-12-05 by gonX