OpenTabletDriver/OpenTabletDriver #3230: Support Request: Flipping pen to use eraser doesn't work as expected on wayland

Zerogaku, 2024-03-09:
### What do you need help with?

I recently managed to get opentabletdriver working on wayland linux, but when flipping the pen to the eraser tip it doesn't activate the eraser brush on krita, I don't know if it is supported on linux because the closest issue I found to my problem is on windows https://github.com/OpenTabletDriver/OpenTabletDriver/issues/3161 and I can't seem to find a "pen tip binding". I do vaguely remember the eraser being activated at one point but it was more of a 1 time switch, so when I flipped back to the pen tip, it didn't swap back to the regular brush, but I must have changed a configuration or something because now it just acts like the normal pen tip. This isn't too bad because I can just press E to activate erase mode on my current brush and change the brush size holding down shift and dragging, but then I have to change back to my previous brush size, so it's not super convenient.

### What have you tried?

Changing eraser binding to linux artist mode button 3

### Diagnostic Information

[diagnostics.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14544906/diagnostics.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

Zerogaku, 2024-03-09:
some screenshots of the current settings
[image]
[image]

Anti-Apple4life, 2024-06-08:
I am having the same issue.

Tablet: CTH-480
My settings:
[image]
[image]

jamesbt365, 2025-01-24:
Works for me on sway, not going to try another implementation as I don't really want to bloat my install, maybe tool reports aren't parsed correctly on this tablet?

diniamo, 2025-01-24:
This was fixed by #3700.