OpenTabletDriver/OpenTabletDriver #3240: Add support for

RavXenz, 2024-03-12:
### Manufacturer

gaomon

### Product Name

pd1220

### Device Width

256.3

### Device Height

144.2

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Diagnostic Information

[gaomonpd1220.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14566393/gaomonpd1220.txt)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-03-12:
That is not a consise or meaningful title, please include the tablets name.

RavXenz, 2024-03-12:
GAOMON PD1220 Pen Display

jamesbt365, 2024-03-12:
Thats a comment not the title of the issue, press the edit button.

RavXenz, 2024-03-12:
my bad. i think i fixed it now

jamesbt365, 2024-03-12:
No you edited the body of your issue, press the edit button by the title and change the title to "Add support for Gaomon PD1220".

RavXenz, 2024-03-12:
ohh im sorry lol i did it