OpenTabletDriver/OpenTabletDriver #3246: InvalidOperationException | Invalid Button '' | Wacom CTC-6110WL

theGeekyLad, 2024-03-13:
### Steps to Reproduce

1. Launch OpenTabletDriver application
2. Set mode to "Artist Mode"
3. Pen Settings -> Pen Binding 1 / 2 / Tip Binding -> "Linux Artist Mode"
4. Apply
5. Save
6. Moving pen around -> all well
7. Press the pen tip down -> `opentabletdriver.service` crashes with an exception
8. `systemctl --user restart opentabletdriver.service --now`
9. Repeat steps 1 - 6
10. Press any pen button -> `opentabletdriver.service` crashes with an exception

### Expected Behavior

The pen tip / buttons should work.

### Observed Behavior

`opentabletdriver.service` crashes (logs below).

### Tablet Device

Wacom CTC-6110WL

### Diagnostic Information

OpenTabletDriver version: 0.6.4.0-3
```
...
Mar 13 01:00:04 tgl-xps otd-daemon[1058608]: [InvalidOperationException:Error]        Invalid Button ''
Mar 13 01:00:04 tgl-xps otd-daemon[1058608]:    at OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding.SetState(Boolean state) in /home/thegeekylad/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.4.0/OpenTabletDriver.Desktop/Binding/LinuxArtistMode/LinuxArtistModeButtonBinding.cs:line 42
Mar 13 01:00:04 tgl-xps otd-daemon[1058608]:    at OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding.Press(TabletReference tablet, IDeviceReport report) in /home/thegeekylad/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.4.0/OpenTabletDriver.Desktop/Binding/LinuxArtistMode/LinuxArtistModeButtonBinding.cs:line 27
Mar 13 01:00:04 tgl-xps otd-daemon[1058608]:    at OpenTabletDriver.Desktop.Binding.BindingState.Invoke(TabletReference tablet, IDeviceReport report, Boolean newState) in /home/thegeekylad/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.4.0/OpenTabletDriver.Desktop/Binding/BindingState.cs:line 25
Mar 13 01:00:04 tgl-xps otd-daemon[1058608]:    at OpenTabletDriver.Desktop.Binding.BindingHandler.HandleBindingCollection(TabletReference tablet, IDeviceReport report, IDictionary`2 bindings, IList`1 newStates) in /home/thegeekylad/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.4.0/OpenTabletDriver.Desktop/Binding/BindingHandler.cs:line 80
Mar 13 01:00:04 tgl-xps otd-daemon[1058608]:    at OpenTabletDriver.Desktop.Binding.BindingHandler.HandleBinding(IDeviceReport report) in /home/thegeekylad/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.4.0/OpenTabletDriver.Desktop/Binding/BindingHandler.cs:line 49
Mar 13 01:00:04 tgl-xps otd-daemon[1058608]:    at OpenTabletDriver.Desktop.Binding.BindingHandler.Consume(IDeviceReport report) in /home/thegeekylad/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.4.0/OpenTabletDriver.Desktop/Binding/BindingHandler.cs:line 38
Mar 13 01:00:04 tgl-xps otd-daemon[1058608]:    at OpenTabletDriver.InputDeviceTree.HandleReport(Object sender, IDeviceReport report) in /home/thegeekylad/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.4.0/OpenTabletDriver/InputDeviceTree.cs:line 65
Mar 13 01:00:04 tgl-xps otd-daemon[1058608]:    at OpenTabletDriver.Devices.DeviceReader`1.Main() in /home/thegeekylad/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.6.4.0/OpenTabletDriver/Devices/DeviceReader.cs:line 111
...
```

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-03-13:
Please upload your settings file from ~/.config/OpenTabletDriver

theGeekyLad, 2024-03-14:
Sure @jamesbt365 , thanks for asking. Here's my `settings.json`:

```
{
  "Profiles": [
    {
      "Tablet": "Wacom CTC-6110WL",
      "OutputMode": {
        "Path": "OpenTabletDriver.Desktop.Output.LinuxArtistMode",
        "Settings": [],
        "Enable": true
      },
      "Filters": [],
      "AbsoluteModeSettings": {
        "Display": {
          "Width": 1440.0,
          "Height": 1440.0,
          "X": 4450.0,
          "Y": 1228.7755,
          "Rotation": 0.0
        },
        "Tablet": {
          "Width": 216.0,
          "Height": 216.0,
          "X": 108.0,
          "Y": 67.5,
          "Rotation": 90.0
        },
        "EnableClipping": true,
        "EnableAreaLimiting": false,
        "LockAspectRatio": false
      },
      "RelativeModeSettings": {
        "XSensitivity": 10.0,
        "YSensitivity": 10.0,
        "RelativeRotation": 0.0,
        "RelativeResetDelay": "00:00:00.1000000"
      },
      "Bindings": {
        "TipActivationThreshold": 0.0,
        "TipButton": null,
        "EraserActivationThreshold": 0.0,
        "EraserButton": null,
        "PenButtons": [
          {
            "Path": "OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding",
            "Settings": [
              {
                "Property": "Button",
                "Value": null
              }
            ],
            "Enable": true
          },
          null
        ],
        "AuxButtons": [],
        "MouseButtons": [],
        "MouseScrollUp": null,
        "MouseScrollDown": null
      }
    }
  ],
  "LockUsableAreaDisplay": true,
  "LockUsableAreaTablet": false,
  "Tools": []
}
```

theGeekyLad, 2024-03-14:
Oh my gosh that totally works. Thanks a billion @jamesbt365 . For posterity, here's my `settings.json` after doing what you asked me to:

```
{
  "Profiles": [
    {
      "Tablet": "Wacom CTC-6110WL",
      "OutputMode": {
        "Path": "OpenTabletDriver.Desktop.Output.LinuxArtistMode",
        "Settings": [],
        "Enable": true
      },
      "Filters": [],
      "AbsoluteModeSettings": {
        "Display": {
          "Width": 3456.0,
          "Height": 2160.0,
          "X": 1728.0,
          "Y": 1080.0,
          "Rotation": 0.0
        },
        "Tablet": {
          "Width": 216.0,
          "Height": 135.0,
          "X": 108.0,
          "Y": 67.5,
          "Rotation": 0.0
        },
        "EnableClipping": true,
        "EnableAreaLimiting": false,
        "LockAspectRatio": false
      },
      "RelativeModeSettings": {
        "XSensitivity": 10.0,
        "YSensitivity": 10.0,
        "RelativeRotation": 0.0,
        "RelativeResetDelay": "00:00:00.1000000"
      },
      "Bindings": {
        "TipActivationThreshold": 0.0,
        "TipButton": {
          "Path": "OpenTabletDriver.Desktop.Binding.MouseBinding",
          "Settings": [
            {
              "Property": "Button",
              "Value": "Left"
            }
          ],
          "Enable": true
        },
        "EraserActivationThreshold": 0.0,
        "EraserButton": null,
        "PenButtons": [
          {
            "Path": "OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding",
            "Settings": [
              {
                "Property": "Button",
                "Value": "Pen Button 1"
              }
            ],
            "Enable": true
          },
          {
            "Path": "OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding",
            "Settings": [
              {
                "Property": "Button",
                "Value": "Pen Button 2"
              }
            ],
            "Enable": true
          }
        ],
        "AuxButtons": [],
        "MouseButtons": [],
        "MouseScrollUp": null,
        "MouseScrollDown": null
      }
    }
  ],
  "LockUsableAreaDisplay": true,
  "LockUsableAreaTablet": true,
  "Tools": []
}
````

Thanks again! 🚀

closed 2024-03-14 by theGeekyLad