OpenTabletDriver/OpenTabletDriver #3248: Wacom PTZ-930 no input

dpark12300, 2024-03-14:
### Steps to Reproduce

Plug in PTZ-930.

### Expected Behavior

Cursor should move onscreen

### Observed Behavior

Nothing happened. Buttons work though.

### Tablet Device

Wacom PTZ-930

### Diagnostic Information

[diag.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14595299/diag.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-03-14:
File > Reset to defaults, save.

dpark12300, 2024-03-14:
Still nothing

jamesbt365, 2024-03-14:
Tablet > tablet debugger, move your pen around on your tablet, do you see a blue dot moving around in the debugger?

dpark12300, 2024-03-14:
No.

jamesbt365, 2024-03-14:
Well that indicates one of 2 things then:
- Any wireless chargers, charging bricks, large speakers, keyboards, mice, phones, but this could also be basically anything electronic device is too close to the tablet
- The pen is dead

The debugger shows the data directly from the tablet, so if you are not seeing any updates then that indicates one of the above.

closed 2024-03-14 by jamesbt365

dpark12300, 2024-03-14:
None of that, and the pen doesn't need batteries.

jamesbt365, 2024-03-14:
Dead in this sense means it doesn't work anymore. No data = no connection between the tablet and pen.

dpark12300, 2024-03-14:
k