OpenTabletDriver/OpenTabletDriver #3252: Daemon not starting after installation on Arch Linux

tarorororororo, 2024-03-16:
### Steps to Reproduce

1. Have an up-to-date Arch Linux installation, possibly with no previous OTD versions installed. As of 3/16/2024, arch 6.7.9.
2. Attempt to install OTD, either from AUR or via makepkg
3. Complete installation
4. Run otd-daemon

### Expected Behavior

otd-daemon runs

### Observed Behavior

```
$ otd-daemon
Failure processing application bundle; possible file corruption.
Arithmetic overflow while reading bundle.
A fatal error occured while processing application bundle

### Tablet Device

N/A

### Diagnostic Information

[Log file of attempted install and run of otd-daemon](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14623678/screen.log)

## Potential fix
1. Clone the pkgbuild from the AUR: 
```git clone https://aur.archlinux.org/opentabletdriver.git```

2. In the PKGBUILD file, using a text editor, change:
``./eng/linux/package.sh -- $EXTRA_OPTIONS``  to ``./eng/linux/package.sh --single-file false -- $EXTRA_OPTIONS``
and replace all lines 65 through 68 with the one line: 
``(PKGBUILD) to cp -r "$_srcdir/dist" "$pkgdir/usr/lib/$_lpkgname"``

      Result should look as follows
      ![image](https://github.com/OpenTabletDriver/OpenTabletDriver/assets/92291273/e2f433b7-e3bf-41f7-b706-267d6f935256)

[PKGBUILD file I used, after editing as above steps 2 and 3, if you would like to replace](https://cdn.discordapp.com/attachments/1164039315341459577/1218550136549146634/PKGBUILD?ex=66081240&is=65f59d40&hm=09d1f8180ee41581faf50da313df3ae1ada66b9346e0f30781ca0b26d7892591&)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

dain98, 2024-03-16:
I had the same issue, potential fix also fixed it.

I'm on EndeavourOS 6.7.9-arch1-1

gabldotink, 2024-03-17:
Same issue, same fix on EndeavourOS 6.8.1-arch1-1. You’re a lifesaver

6b6279, 2024-03-18:
This made otd-daemon not crash upon startup, but OTD can't detect my Wacom CTL-4100 tablet despite the matching udev rule and the device being correctly connected (according to lsusb). Could this be due to some missing file?

Javatrix, 2024-03-18:
How did you even find it? Thank you, I'm so lucky I found your post.

gonX, 2024-03-19:
For anyone else experiencing the issue, does it fix the issue if you change the `options` array in the PKGBUILD to `options=('!strip' 'staticlibs')`?

gabldotink, 2024-03-21:
> For anyone else experiencing the issue, does it fix the issue if you change the `options` array in the PKGBUILD to `options=('!strip' 'staticlibs')`?

Doesn’t fix for me.

jamesbt365, 2024-03-21:
capitalize OPTIONS and it'll go away, probably, something changed in packaging upstream and now lowercase doesn't work for some people.

gabldotink, 2024-03-21:
> capitalize OPTIONS and it'll go away, probably, something changed in packaging upstream and now lowercase doesn't work for some people.

This works

jamesbt365, 2024-03-21:
A hotfix has been pushed to the AUR that circumvents the upstream issue [here](https://gitlab.archlinux.org/archlinux/packaging/packages/pacman/-/issues/19).

closed 2024-03-21 by jamesbt365